semuconsulting/PyGPSClient #1: Make console output copyable

lenhart, 2021-03-18:
Hey,
first: thanks for the great work!

I was wondering if the text console could be made copyable (i.e. I cannot copy text on Ubuntu Linux 20.04). 
I understand that it comes from the _disabled_ state to make the text field non-modifiable, there seem to be a couple work-arounds for this problem (see SO links below), but as I have never used tkinter before and am no python expert, I am not really sure what would be a good way to do this. I am happy to test and try stuff out, but before sinking in a lot of time and work for that feature I thought I ask you first (otherwise my local hotfix would be to set the text back to normal behavior..)

https://stackoverflow.com/questions/1602106/in-pythons-tkinter-how-can-i-make-a-label-such-that-you-can-select-the-text-wi
https://stackoverflow.com/questions/3842155/is-there-a-way-to-make-the-tkinter-text-widget-read-only#3842234

Best regards
Malte

semuadmin, 2021-03-18:
Hi - thanks for your interest. Not sure what particular use case you have in mind, but a few options come to mind that don't involve having to mess around with the text box settings.

1. Obviously if you just want to copy and paste raw (unparsed) output, you can use PyGPSClient's existing datalogging facility. The datalogging facility doesn't currently output parsed data, but I have been meaning to add this as an option at some point. If such an option was available, would that meet your requirement?
2. In the short term, the quickest and easiest solution might be to use the pyubx2 and/or pynmeagps command line parsers and simply copy the parsed output to a logfile using your terminal's native logging facilities (e.g. screen or PuTTY). Both libraries have a simple 'dump' utility in their respective examples folders that you could use for this purpose. This way you can copy and paste as much output as you like, rather than be limited to the contents of the text box. [pynmeagps nmeadump](https://github.com/semuconsulting/pynmeagps/tree/master/examples) [pyubx2 ubxdump](https://github.com/semuconsulting/pyubx2/tree/master/examples).

lenhart, 2021-03-18:
Hey, thanks for the reply! 

My use case was to copy paste individual messages to include in a work log or as a comment into the code, so I'd have a reference which fields are available in that particular message.
I managed to do that with pyubx2 which I am using mainly to do some parsing (thank also for that!), but I thought it might be practical to copy things from the gui as well. 
My intention was to check with you how much work that would be, if there exist a clean enough solution or if you had an idea on that (noticed that I didn't make that too clear in the initial message).

Logging of parsed messages would sort of solve the issue, but still would a lot more work if you just wanted to look up a specific message.. but I also see the point that this might be a rare use-case. My initial thought was "how hard can it be to fix that", but that did not turn out to be too true. The _readonly_ property unfortunately does not exist for that class.

Thank you for your suggestions! I guess we can lay it aside now. If more people request it at least you have a few references :)
Cheers!

semuadmin, 2021-03-18:
It's a good suggestion. I'm thinking that adding an option to datalog parsed output as well as raw output (which is relatively trivial to do) would probably be the most flexible solution. As I say, it affords you unlimited output to copy and paste from, plus you can use your preferred text / hex editor to search for any particular entries you're after. Are you happy for me to log that as a feature request for the next release and close this issue?

lenhart, 2021-03-18:
Thank you! As there is no easy/clean solution to copy from the console tile, yes that would be the best alternative!
You are welcome to close the issue, thanks for your prompt replies :+1: 

When you are working on the output parts, I noticed that when checking the log or record gpx file boxes and then closing the dialog w/o choosing a path, some exception will be propagated to the terminal which started pygpsclient. Maybe that could be caught and silenced somewhere as it is not really relevant I think :-) )

lenhart, 2021-03-18:
I attach some more feedback into this issue.. if you rather want it separately I can o.c. open more issues, but some of this is really small..

feature enhancements:
- on startup: if no known device was recognised (happens to me under Linux), I think it would be nice to pre-select the first from the list instead of leaving it without selection (because the first one when connecting a ublox receiver is the right one)
- when started: a button to refresh the port list would be nice (or detecting plug events and automatically doing so, don't know if that is feasible). That way one wouldn't have to restart the program when plugging in a new receiver. (From own experience writing a multiple serial device program - in Java; don't know if python has similar limitations or not - I directly suggest to disable the button in a connected state or keep track of the currently connected ones and don't replace the ones in the list to not loose the serial port handler to close it later :-) )

minor gui enhancements:
- the ubx config dialog from the menu is clickable before connecting to a device but no longer after disconnecting
- does tkinter have pop-up hints for buttons/checkboxes? It took me a while to figure out when the gpx/logging boxes are selectable or not. maybe a hint would do (also is there a reason to not allow starting a recording after connecting?)
- hints in general could be beneficial for some users. When writing this feedback list I noticed the button between connect and disconnect. Completely masked that out in my brain because I didn't know what it was for. Loading a config file I suppose? (after stumbling over such a feature when looking through the code for how the serial list is created on startup)
- are there two blanks on the label "Web Map Zoom"  between map & zoom? 

cheers!

semuadmin, 2021-03-18:
Re. context sensitive hints - this can be done in tkinter but it's hugely cumbersome in code and I didn't feel it was warranted here - I've endeavoured to document the functionality fairly comprehensively in the README. The blue icon button next to the red icon Connect button is for streaming from a file (e.g. a saved datalog) - this is covered in the README. 
Dynamically updating serial ports is somewhat more problematic using the comports library in Python than with the equivalent in Java or C++, but it is possible. Other points noted - thanks for the feedback; appreciated.

semuadmin, 2021-03-19:
Issues should be addressed in release candidate 1.0.3.

lenhart, 2021-03-19:
Hey, looks very good! I am impressed how fast you managed to implement all this!
(I wasn't even able to provoke an error when deleting the logfile while it tried writing to it :D)

New comments:
- the auto select first serial port feature would be more obvious if the selected port also was highlighted in the gui. If that is simple I'd suggest it, ow this works well enough :-)
- I would still argue that the error msg "Map not available" is not really describing the cause of the error. 'Map disabled. No position fix' would for instance describe it better in my opinion. But in the end it is your choice :-)

One error poped up a couple times, apparently when parsing a certain msg. Maybe you can make sense of it
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "/home/malte/software/PyGPSClient/pygpsclient/serial_handler.py", line 304, in on_read
    self._parse_data(self._serial_buffer)
  File "/home/malte/software/PyGPSClient/pygpsclient/serial_handler.py", line 366, in _parse_data
    parsed_data = self.__app.ubx_handler.process_data(raw_data)
  File "/home/malte/software/PyGPSClient/pygpsclient/ubx_handler.py", line 100, in process_data
    self._process_MON_HW(parsed_data)
  File "/home/malte/software/PyGPSClient/pygpsclient/ubx_handler.py", line 520, in _process_MON_HW
    self.__app.dlg_ubxconfig.update_pending(
  File "/home/malte/software/PyGPSClient/pygpsclient/ubx_config_dialog.py", line 239, in update_pending
    self.set_status(
  File "/home/malte/software/PyGPSClient/pygpsclient/ubx_config_dialog.py", line 268, in set_status
    self._lbl_status.config(fg=color)
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1637, in configure
    return self._configure('configure', cnf, kw)
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1627, in _configure
    self.tk.call(_flatten((self._w, cmd)) + self._options(cnf))
_tkinter.TclError: invalid command name ".!toplevel.!frame.!frame.!label"
```

Meanwhile I actually found a way to copy from the console view. Selecting text and pressing the middle mouse button copies it, a new press somewhere else inserts it. I think that is unix only behaviour though. Funnily ctrl + c also works to copy, but again middle mouse button is the only way to insert.

So thanks for all the work and a good weekend!

semuadmin, 2021-03-19:
Ok few quick responses for now...
1) AFAIK tkinter doesn't provide any consistent way to highlight selected entries in Listboxes across different platforms. e.g. on Windows the selection is underlined, but not on MacOS. The bottom status bar does tell you which device you're connected to so I don't see this as a major issue.
2) The 'no signal' advisory message on the map panel says "Map Not Available. No Satellite Fix." - is that not fairly explicit?
3) I'm unable to recreate the error you've reported with MON-HW messages on Ubuntu 20.4 or any platform or device - can you provide any additional details e.g. what version of tkinter do you have installed? what specific GNSS device and firmware level are you using? Any other context - under what circumstances do you see the error? Are you able to recreate it at will - if so, how? Cheers.

lenhart, 2021-03-22:
Hey, 
1. selected port is now easy visible and obvious on Ubuntu 20.04 (good that it took me a few days to answer, good work :-) )
2. my thought was, that it is not really the map that is unavailable. It is just chosen to disable it without a position fix. But that is very minor and one should be able to understand the meaning in any case (a bit interpretation thing and maybe it is just me who thinks that it could be misread)
3. error unfortunately not replicable at the moment.. just showed up a few times and I could not make sense of it, so I thought it might be something obvious to you... but now that I think of it, I had a few problems with the receiver because I had too many messages activated and a few were dropped in the resulting congestion. That could have of course led to msg corruptions as well.. If I see it again I'll try to get more details. So, nothing to worry more about for now I think (but to answer the questions: tkinter 8.6, ublox F9P gnss-receiver)

semuadmin, 2021-03-23:
Closed with v1.0.3

closed 2021-03-23 by semuadmin