semuconsulting/PyGPSClient #12: MapQuest API key via env variable

dizcza, 2022-02-03:
Hi and thanks for a unix-friendly UI to parse ublox data. I was looking for a non-Windows solution and found yours.

I have lots of files in a home directory, and typically the keys are set via environment variables. Could you add such an option to parse the `MQAPIKEY` env var, if set, as an alternative to the `mqapikey` file?

Also, is there a way to show the data in Google Maps?

Thanks.

dizcza, 2022-02-03:
Also, I don't understand what the `track` and `speed` are in UI parsed data. Other tags are self-explanatory. Could you perhaps add some explanation for these two variables? For example, I always get N/A for the `track` and I don't know how bad it is.

semuadmin, 2022-02-03:
Hi @dizcza ,

Thanks for your interest in PyGPSClient - glad you're finding it useful.

To answer your questions in order:

1. There are no current plans to enable MQAPIKEY via an environment variable - this can require elevated privileges on some platforms which makes it more fiddly to implement on a consistent cross-platform basis. Hopefully the key file option can work for you.
2. Simple answer is no - I'm using the MapQuest API because it's free (for non-commercial use up to 15,000 transactions a month) and relatively easy to implement. Out of interest, do you see any particular advantage in using Google Maps rather than MapQuest? The maps are equally reliable in my experience.
3. Can you clarify which particular UBX or NMEA message type(s) you're referring to containing the `track` and `speed` attributes? Without knowing which device you're using or how/where it's configured, I can't speak for the particular readings you're getting, but if you tell me which message type you're getting it from I can at least advise on its interpretation.

dizcza, 2022-02-03:
I'm using a Neo-6M.

semuadmin, 2022-02-03:
ok that's a fairly old device. which specific NMEA message type are you getting `speed` and `track` from?

dizcza, 2022-02-03:
>  which specific NMEA message type are you getting speed and track from?

I don't know. That's the parsed log I see in your app:

```
<NMEA(GPRMC, time=15:40:26, status=A, lat=50.4648645, NS=N, lon=30.48573683, EW=E, spd=0.657, cog=, date=2022-02-03, mv=, mvEW=, posMode=A)>
<NMEA(GPVTG, cogt=, cogtUnit=T, cogm=, cogmUnit=M, sogn=0.657, sognUnit=N, sogk=1.217, sogkUnit=K, posMode=A)>
<NMEA(GPGGA, time=15:40:26, lat=50.4648645, NS=N, lon=30.48573683, EW=E, quality=1, numSV=4, HDOP=3.35, alt=153.7, altUnit=M, sep=25.7, sepUnit=M, diffAge=, diffStation=)>
<NMEA(GPGSA, opMode=A, navMode=3, svid_01=29, svid_02=25, svid_03=2, svid_04=12, svid_05=, svid_06=, svid_07=, svid_08=, svid_09=, svid_10=, svid_11=, svid_12=, PDOP=7.66, HDOP=3.35, VDOP=6.89)>
<NMEA(GPGSV, numMsg=4, msgNum=1, numSV=15, svid_01=2, elv_01=52.0, az_01=70, cno_01=27, svid_02=5, elv_02=7.0, az_02=128, cno_02=, svid_03=6, elv_03=14.0, az_03=38, cno_03=, svid_04=11, elv_04=47.0, az_04=68, cno_04=25)>
<NMEA(GPGSV, numMsg=4, msgNum=2, numSV=15, svid_01=12, elv_01=51.0, az_01=114, cno_01=46, svid_02=18, elv_02=1.0, az_02=200, cno_02=25, svid_03=20, elv_03=20.0, az_03=103, cno_03=, svid_04=22, elv_04=11.0, az_04=276, cno_04=22)>
<NMEA(GPGSV, numMsg=4, msgNum=3, numSV=15, svid_01=24, elv_01=4.0, az_01=173, cno_01=34, svid_02=25, elv_02=87.0, az_02=258, cno_02=25, svid_03=26, elv_03=0.0, az_03=299, cno_03=, svid_04=29, elv_04=58.0, az_04=251, cno_04=33)>
<NMEA(GPGSV, numMsg=4, msgNum=4, numSV=15, svid_01=31, elv_01=34.0, az_01=303, cno_01=, svid_02=32, elv_02=3.0, az_02=258, cno_02=31, svid_03=39, elv_03=32.0, az_03=187, cno_03=38)>
<NMEA(GPGLL, lat=50.4648645, NS=N, lon=30.48573683, EW=E, time=15:40:26, status=A, posMode=A)>
<NMEA(GPRMC, time=15:40:27, status=A, lat=50.46486117, NS=N, lon=30.48573433, EW=E, spd=0.461, cog=, date=2022-02-03, mv=, mvEW=, posMode=A)>
<NMEA(GPVTG, cogt=, cogtUnit=T, cogm=, cogmUnit=M, sogn=0.461, sognUnit=N, sogk=0.854, sogkUnit=K, posMode=A)>
<NMEA(GPGGA, time=15:40:27, lat=50.46486117, NS=N, lon=30.48573433, EW=E, quality=1, numSV=4, HDOP=3.35, alt=154.9, altUnit=M, sep=25.7, sepUnit=M, diffAge=, diffStation=)>
<NMEA(GPGSA, opMode=A, navMode=3, svid_01=29, svid_02=25, svid_03=2, svid_04=12, svid_05=, svid_06=, svid_07=, svid_08=, svid_09=, svid_10=, svid_11=, svid_12=, PDOP=7.66, HDOP=3.35, VDOP=6.89)>
<NMEA(GPGSV, numMsg=4, msgNum=1, numSV=15, svid_01=2, elv_01=52.0, az_01=70, cno_01=27, svid_02=5, elv_02=7.0, az_02=128, cno_02=, svid_03=6, elv_03=14.0, az_03=38, cno_03=, svid_04=11, elv_04=47.0, az_04=68, cno_04=25)>
<NMEA(GPGSV, numMsg=4, msgNum=2, numSV=15, svid_01=12, elv_01=51.0, az_01=114, cno_01=46, svid_02=18, elv_02=1.0, az_02=200, cno_02=26, svid_03=20, elv_03=20.0, az_03=103, cno_03=, svid_04=22, elv_04=11.0, az_04=276, cno_04=23)>
<NMEA(GPGSV, numMsg=4, msgNum=3, numSV=15, svid_01=24, elv_01=4.0, az_01=173, cno_01=34, svid_02=25, elv_02=87.0, az_02=258, cno_02=25, svid_03=26, elv_03=0.0, az_03=299, cno_03=, svid_04=29, elv_04=58.0, az_04=251, cno_04=33)>
<NMEA(GPGSV, numMsg=4, msgNum=4, numSV=15, svid_01=31, elv_01=34.0, az_01=303, cno_01=, svid_02=32, elv_02=3.0, az_02=258, cno_02=31, svid_03=39, elv_03=32.0, az_03=187, cno_03=38)>
<NMEA(GPGLL, lat=50.46486117, NS=N, lon=30.48573433, EW=E, time=15:40:27, status=A, posMode=A)>
<NMEA(GPRMC, time=15:40:28, status=A, lat=50.46485033, NS=N, lon=30.48572967, EW=E, spd=0.198, cog=, date=2022-02-03, mv=, mvEW=, posMode=A)>
<NMEA(GPVTG, cogt=, cogtUnit=T, cogm=, cogmUnit=M, sogn=0.198, sognUnit=N, sogk=0.366, sogkUnit=K, posMode=A)>
<NMEA(GPGGA, time=15:40:28, lat=50.46485033, NS=N, lon=30.48572967, EW=E, quality=1, numSV=4, HDOP=3.35, alt=156.1, altUnit=M, sep=25.7, sepUnit=M, diffAge=, diffStation=)>
<NMEA(GPGSA, opMode=A, navMode=3, svid_01=29, svid_02=25, svid_03=2, svid_04=12, svid_05=, svid_06=, svid_07=, svid_08=, svid_09=, svid_10=, svid_11=, svid_12=, PDOP=7.66, HDOP=3.35, VDOP=6.89)>
<NMEA(GPGSV, numMsg=4, msgNum=1, numSV=15, svid_01=2, elv_01=52.0, az_01=70, cno_01=27, svid_02=5, elv_02=7.0, az_02=128, cno_02=, svid_03=6, elv_03=14.0, az_03=38, cno_03=, svid_04=11, elv_04=47.0, az_04=68, cno_04=25)>
<NMEA(GPGSV, numMsg=4, msgNum=2, numSV=15, svid_01=12, elv_01=51.0, az_01=114, cno_01=46, svid_02=18, elv_02=1.0, az_02=200, cno_02=26, svid_03=20, elv_03=20.0, az_03=103, cno_03=, svid_04=22, elv_04=11.0, az_04=276, cno_04=24)>
<NMEA(GPGSV, numMsg=4, msgNum=3, numSV=15, svid_01=24, elv_01=4.0, az_01=173, cno_01=34, svid_02=25, elv_02=87.0, az_02=258, cno_02=25, svid_03=26, elv_03=0.0, az_03=299, cno_03=, svid_04=29, elv_04=58.0, az_04=251, cno_04=33)>
<NMEA(GPGSV, numMsg=4, msgNum=4, numSV=15, svid_01=31, elv_01=34.0, az_01=303, cno_01=, svid_02=32, elv_02=3.0, az_02=258, cno_02=31, svid_03=39, elv_03=32.0, az_03=187, cno_03=38)>
<NMEA(GPGLL, lat=50.46485033, NS=N, lon=30.48572967, EW=E, time=15:40:28, status=A, posMode=A)>
```

dizcza, 2022-02-03:
> do you see any particular advantage in using Google Maps rather than MapQuest?

I can freely navigate in a google map.

dizcza, 2022-02-03:
> this can require elevated privileges on some platforms which makes it more fiddly to implement on a consistent cross-platform basis.

Why? The `os` built-in module is cross-platform, isn't it?

```python
import os
key = os.environ.get("MQAPIKEY")
```

semuadmin, 2022-02-03:
OK so the log you've provided contains a variety of NMEA messages - the message type is bit after `<NMEA(,` so for example `GPRMC`, `GPVTG`, `GPGLL` etc. (_CAUTION - the coordinates in this log are now visible in the public domain_)

The `spd` attribute in the GPRMC message, for example, represents the speed over ground in knots; likewise the `sogn`  attribute in GPVTG. The `sogk` attribute in GPVTG represents speed over ground in km/h. I would refer you to the data sheet for your particular device for further information, or post a general question in the u-blox support forum https://portal.u-blox.com/s/.

dizcza, 2022-02-03:
> CAUTION - the coordinates in this log are now visible in the public domain

True. I'll delete the log.

You're right about sticking to simplicity. Your app is cool anyway and perhaps it got working from the start just because it's simple, meant to have a quick look, and lacks features that otherwise would turn your project into a gigantic dependencies list (which would make me feel to search for a simpler solution).

dizcza, 2022-02-03:
All right. Thanks for your help.

closed 2022-02-03 by dizcza

semuadmin, 2022-02-03:
> All right. Thanks for your help.

More than happy to consider feature requests. I will as I say look into provisioning map keys as ENV variables, but I may not get round to it for a while. In the meantime, if there's any functionality you think `PyGPSClient` could usefully provide _within the context of_ a (relatively) simple Python/tkinter diagnostic and configuration tool, by all means raise the suggestion. Cheers.

dizcza, 2022-02-03:
Since you asked, I actually do have feedback to share. If I understood correctly, you encode the satellite type with color. But then you also encode the satellite signal with color. Both ideas are great but put together, they might confuse the users. It took me quite some time to map the color of satellite bars (middle plot, header bars: GPS, SBA, GAL,...) with the color of satellite circles via integer ID values. I have two suggestions. You can pick any, both, or none of them.

1. [Middle panel] Make the satellite names clickable. Boolean type. Either selected or not. Basically, a filter. For example, I want to select GAL only. Then I deselect except GAL. Clicking on the satellite name box (rectangle) again will make it active again (bring it back onto the map - both left and middle panels).
2. [Left panel] Hovering on the satellite circle will show the satellite info. At least, the satellite type. Because I still don't understand what types of satellites are visible in my GPS module.

[Middle panel] Alternatively to both of them, you could make opaque those satellite box names that are not captured by the device. No user interactions (read simple), as you currently have. But this will prove useful to know which satellite types are currently present.

semuadmin, 2022-02-03:
Hi again. Thanks for the feedback - appreciated. The colour coding was an attempt to represent both the GNSS constellation type and the signal level at the same time. Ironically I notice that u-blox themselves have adopted a very similar scheme in the latest u-center 2! The colours themselves are configurable (in `globals.py`) - I did agonise over the colours used, both in terms of accessibility (colour blindness etc.) and 'political correctness', though in the end, red for GLONASS did seem a logical choice ;-).  I could provide a switch to colorise _either_ the signal strength _or_ GNSS constellation but not both at the same time - I'll think about it...

Not entirely clear about your last paragraph - the display only shows those satellites supported (or 'captured') by your device at any time - obviously these will change as your location and local time changes. There is already an option to filter out those satellites which are 'seen' by your device but not used in the navigation solution.

You can filter the various different GNSS constellations using CFG commands (_I provide some examples in my UBXPRESETS template_), but a quick filter option is not a bad idea and fairly easily done, so happy to consider that (u-center does something similar). The satellite numbering is configurable to some extent (there are some options in `globals.py`), but there's no definitive numbering scheme across all GNSS constellations so the choice is arbitary to some extent.

Implementing 'tooltip' style hover-over popups is somewhat fiddly in tkinter so I've avoided that to date, though I appreciate it can offer a nice user experience (it's much easier to do in, say, Qt or Electron / Node.js). To be honest, tkinter appears to be a bit broken in Python 3.10 on MacOS Monterey at the moment so I'm holding off doing any significant development work in tkinter until I'm confident Apple remain committed to supporting it. It still works fine on Windows and most Linux distros though.

dizcza, 2022-02-03:
> Implementing 'tooltip' style hover-over popups is somewhat fiddly in tkinter so I've avoided that to date

Hovering over individual satellites I'm considering the least valuable option.

> Not entirely clear about your last paragraph - the display only shows those satellites supported (or 'captured') by your device at any time - obviously these will change as your location and local time changes.

I mean, you've fixed all possible satellite types in the header of the middle plot. I call them boxes. The picture below shows that my GPS module has captured no good signal, no satellites available (this is because I'm sitting in a room without an antenna attached). To further show that no satellite is captured, make all boxes opaque (grayed out slightly). Then, for example, one GAL satellite appeared on the horizon with a good signal -> make GAL box non-opaque. In the picture below, all boxes are non-opaque as if all satellites are currently present (while in fact none of them is).

[image]

> There is already an option to filter out those satellites which are 'seen' by your device but not used in the navigation solution.

I didn't understand you. Is there such an option already in your UI app?

> The colours themselves are configurable (in globals.py) 

Any color palette is OK for me. The colors themselves - that's not what I meant.

> I could provide a switch to colorise either the signal strength or GNSS constellation but not both at the same time - I'll think about it...

That could harden the user experience (it'll be slightly heavier in UI with little gain in the functionality).

> The satellite numbering is configurable to some extent (there are some options in globals.py), but there's no definitive numbering scheme across all GNSS constellations so the choice is arbitary to some extent.

The numbering is fine for me.

semuadmin, 2022-02-03:
> To further show that no satellite is captured, make all boxes opaque (grayed out slightly). 

Hmmm - apologies if I'm getting the wrong end of the stick, but AFAICS the only way that would work is if I retain a slot in the graph for every possible GNSS satellite and then only 'colorise' the ones that are 'visible' (or contributing to the navigation solution) at that time. Given there are well over 100 operational GNSS satellites, only a small proportion of which would be visible from a given ground location at any time, that would mean an awful lot of 'grayed out' spaces in the graph. Not sure that's practical. There are in fact no 'reserved' slots in my graph - your screenshot above actually indicates that there are something like 22 satellites 'present' and 'visible' to the device but they all have a very low signal-to-noise ratio and are not therefore contributing to the navigation solution. You could filter these out by unchecking the 'Show Unused Satellites' box.

dizcza, 2022-02-03:
Ah OK, then I misunderstood the program output.
Sure, it won't make sense to plot 100 of them. I thought that there are only limited (<10) types of satellites. I've probably read about the most used/available satellites to the end-users, and I wasn't aware that there are actually on an order of magnitude more.

My misunderstanding came from the fact that when the "Show unused satellites" is not selected, I see none in the left panel but still see all these boxes above in the middle panel. That is how I thought there are only 7 possible satellite types out there flying around the Earth. How wrong could I be.

Then filtering by the satellite type that are captured by clicking on the boxes will be a useful feature. Not really super useful, but something you may consider doing in the outbreak time.

semuadmin, 2022-02-03:
last one for now ...  the 'boxes above' in the middle panel are simply a legend for the colour coding - they don't signify the presence or absence of a particular satellite or constellation and - again - you can turn them off by unchecking the 'Show Legend' box. 

Note, however, noticed that this only takes effect when the widget is refreshed - it was done that way for a reason, but I could probably improve the user experience a bit.

PR #13 (2022-02-06): Load mqapikey from env variable -  Fixes #12
# PyGPSClient Pull Request Template

## Description

Add the ability to load a MapQuest API key from an environment variable as well as a file.

Fixes #12

## Testing

Please test all changes, however trivial, against the supplied unittest suite `tests/test_*.py` e.g. by executing the `tests/testsuite.py` module or using your IDE's native Python unittest integration facilities.
Please describe any test cases you have amended or added to this suite to maintain >= 99% code coverage.

- [x] Tested on Windows, MacOS, LInux and RaspiOS

## Checklist:

-  [... cut]
commit e4fba691 (2022-02-06): Merge pull request #13 from semuconsulting/load-MQAPIKEY-from-ENV-variable

Load mqapikey from env variable -  Fixes #12