semuconsulting/PyGPSClient #18: Support TCP connection to ublox receiver

ArrEssJay, 2022-04-13:
**Is your feature request related to a problem? Please describe.**

Can't configure connection to a remote ublox receiver via TCP.

**Describe the solution you'd like**

Allow configuration of communication via a TCP socket in addition to serial. In this case a remote F9P receiver is connected to a headless server and `socat` used to bridge UART<->TCP.

**Describe alternatives you've considered**

U-Center can use a TCP socket for communication with the receiver, but is Windows only.

**Would you be willing to contribute a test device?**

Yes, as above.

semuadmin, 2022-04-14:
Hi @ArrEssJay I have been toying with the idea of adding functionality along these lines but can give no guarantees on timeframe at this point - PyGPSClient is maintained entirely by unpaid volunteers.

I'd need to clarify what sort of security protocols would need to be implemented as there would clearly be risks associated with opening unauthenticated TCP sockets at distance via socat or similar. I'll put it on the backlog.

If you could share the details of your server side socat configuration (via email if you prefer) that might save some time. Cheers.

gdt, 2022-04-14:
I also think this would be useful.   Besides a regular computer interfacing to a receiver, there are devices like the Ardusimple F9P boards with the WiFi NTRIP master ESP32-based board, which provide a TCP socket that one can obtain data from.  When using a cellphone for RTCM , the Ardusimple device is a hotspot client, and there are no other connected devices.

I see your point about security, but whether a TCP connection is a problem or isn't a problem depends on many things, and I don't think it makes sense to decline to implement a feature because someone might do something.   After all, someone could configure the client to publish the data on the web, which would a location confidentiality violation.    I'm saying this even though I am very much on the paranoid side.

semuadmin, 2022-04-14:
Agreed and on it. Just can't say when... 

Priority at moment is trying to fix/work around tkinter issues on MacOS.

Watch this space. Appreciate the interest.

semuadmin, 2022-04-14:
As an aside in relation to provisioning GNSS data via TCP socket, I also have a project on the back-burner to write a new version of PyGPSClient as an Electron (i.e Node.js) application, retaining the Python parsers but utilising TCP to feed the parsed data to the HTML/JavaScript GUI. Different use case to this issue but maybe some synergies.

ArrEssJay, 2022-04-21:
Slight mistake at my end. A typical way to do this is to use ser2net to act
as a terminal server, and socat to create a virtual serial port.

With ser2net, a Ublox receiver on a USB port could be mapped to TCP port
2000:
$ ser2net -C "localhost,2000:raw:400:/dev/ttyACM0:115200 8DATABITS NONE
1STOPBIT" -d

In U-Center, this receiver could be accessed at localhost:2000

Alternatively, we can use socat to map a virtual serial port to a TCP
socket:
$ socat pty,link=$HOME/dev/ttyV0,waitslave tcp:localhost:2000

Given that is possible,  I don't think this needs to be considered a high
priority issue.

Yes, there is a security issue in the above examples. I certainly would not
recommend using that example on a public network. TLS is implemented in
ser2net:
https://man.archlinux.org/man/community/ser2net/ser2net.yaml.5.en

semuadmin, 2022-04-21:
> Slight mistake at my end. A typical way to do this is to use ser2net to act as a terminal server, and socat to create a virtual serial port. With ser2net, a Ublox receiver on a USB port could be [...]

Sure - `ser2net` / `socat` would be the standard way to do this on Posix platforms, but they're not the easiest tools to get to grips with if you're not a cli power user. Implementing a simple (unauthenticated) socket client along the lines of the Network Connection facility in u-center 21.9 is not particularly difficult - as I say, the requisite code is pretty much already there for the new NTRIP client - but it would obviously carry a 'use at your own risk' warning _(implicit in the BSD-2 licence anyway)_ especially - as you say - if you take it outside your private LAN.

closed 2022-04-28 by semuadmin

ArrEssJay, 2022-05-04:
Nice! Confirmed working.

PR #21 (2022-04-28): Add socket reader and enhance datalogging - Fixes #18 Fixes #19
# PyGPSClient Pull Request Template

## Description

New BETA functionality to support reading NMEA, UBX and RTCM3 datastreams from Socket (TCP/UDP) in addition to Serial (USB/UART) or Binary Datalog data streams. This BETA version supports open (i.e. unuthenticated, unencrypted) sockets, but may be enhanced in future versions.

Introduces new `socket_handler.py` and `socketconfig_frame.py` class modules.

Also incorporates enhancements to datalogging to record incoming NTRIP RTCM3 data stream in addition to date coming from the r [... cut]
commit dacd5dc6 (2022-04-28): Merge pull request #21 from semuconsulting/add-socket-reader

Add socket reader and enhance datalogging - Fixes #18 Fixes #19