semuconsulting/PyGPSClient #19: Saving data to a file for later analysis

foxittt, 2022-04-25:
Hello

Is it possible to implement the saving of data streams(UBX, NMEA, RTCM) in the appropriate files for further analysis.

semuadmin, 2022-04-25:
Hi @foxittt 

PyGPSClient already supports datalogging:

[image]

Simply click the 'Enable Datalog' checkbox _prior to_ connecting to your receiver and select the required format (binary, parsed, hex string or tabular hex). You'll be prompted to select the folder into which the timestamped datalog files are saved. If you wish to replay the datalog within PyGPSClient, choose the binary format.

If you only want to log a particular protocol (e.g. UBX or RTCM), simply de-select the protocols you don't want to see using the 'Protocols Displayed' checkboxes:

[image]

Hope this helps (this information is all in the README, by the way).

foxittt, 2022-04-25:
Thanks for answer.
Trying to log rtcm stream from ntripserver.
But there is the mistake.

semuadmin, 2022-04-25:
For what purpose, out of interest? What 'further analysis' would you be looking to undertake?

foxittt, 2022-04-25:
For example, how different parameters change during the day: the number of visible satellites, how a certain location changes compared to the true and compared to the average.

semuadmin, 2022-04-25:
OK fair enough. It's simple enough to include the incoming NTRIP data in the datalog file. I'll put it on the backlog.

Note that DGPS corrections will only be applied to a connected receiver from a LIVE NTRIP data stream, not from a pre-recorded datalog file.

semuadmin, 2022-04-25:
@foxittt I've uploaded a branch https://github.com/semuconsulting/PyGPSClient/tree/enhance-datalogging which contains enhancements to support NTRIP datalogging. If you want to install and try this version out and let me know how you get on, that would be very helpful.

foxittt, 2022-04-25:
Thanks!!!
I`ll try!

PR #21 (2022-04-28): Add socket reader and enhance datalogging - Fixes #18 Fixes #19
# PyGPSClient Pull Request Template

## Description

New BETA functionality to support reading NMEA, UBX and RTCM3 datastreams from Socket (TCP/UDP) in addition to Serial (USB/UART) or Binary Datalog data streams. This BETA version supports open (i.e. unuthenticated, unencrypted) sockets, but may be enhanced in future versions.

Introduces new `socket_handler.py` and `socketconfig_frame.py` class modules.

Also incorporates enhancements to datalogging to record incoming NTRIP RTCM3 data stream in addition to date coming from the r [... cut]
commit dacd5dc6 (2022-04-28): Merge pull request #21 from semuconsulting/add-socket-reader

Add socket reader and enhance datalogging - Fixes #18 Fixes #19
closed 2022-04-28 by semuadmin