semuconsulting/PyGPSClient #33: Exporting parsed UBX data

fener1efes2, 2022-09-25:
I have created a .ubx data by using u-center.
PyGPSClient succesfully parsed the data.
But I want to export the data to a txt so that I can depict a graph of certain parameters.
I tried to copy from the console but it did not work. Apart form CTRL + C, CTRL + F does not work in the console.
Since it is not useful to look at the data by using console, there must be a property where users can copy or export the data from the console.

[image]

fener1efes2, 2022-09-25:
Data Logging exports the parsed data

closed 2022-09-25 by fener1efes2

semuadmin, 2022-09-26:
Hi @fener1efes2

Yes, that's the solution I recommend. You can also log the hexadecimal alongside the parsed data for a side-by-side comparison.

FYI if you want a non-GUI way of doing the same thing, the `gnssdump` CLI utilility installed with the `pygnssutils` library also offers configurable extract and logging formats. Type `gnssdump -h` for a full set of parameters.

https://github.com/semuconsulting/pygnssutils#gnssdump