semuconsulting/PyGPSClient #36: Configuration Save to File

AgileEngineeringLLC, 2022-12-24:
**Is your feature request related to a problem? Please describe.**
no
A clear and concise description of what the problem is e.g. I'd like to be able to do this [...]
I'd like to be able to save the configuration to a file, and then be able to load that configuration back later (or to a second device)

**Describe the solution you'd like**
I'd use the current menus to configure the device (I use F9Ps), and then I'd like to store that configuration to a file. This would allow me to quickly configure that same device or another device to my saved configuration. Thx!

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

**Would you be willing to contribute a test device?**

If the request relates to a specific u-blox device that is not currently supported, would you be 
willing to contribute a device to the project for testing purposes?

**Additional context**

Add any other context about the feature request here.

semuadmin, 2022-12-26:
Hi @AgileEngineeringLLC,

This is not the first time I’ve had a request for such a feature, but unfortunately there is no trivial solution here. To the best of my knowledge, even u-blox’s own u-center tool does not support this facility.

PyGPSClient does already partially support this requirement via its [User-Defined Preset](https://github.com/semuconsulting/PyGPSClient#userdefined) facility, which allows users to store their own sequences of UBX queries or commands in a `ubxpresets` file and send these to any number of compatible devices via a single mouse-click. However, this does require a certain amount of ‘hand-cranking’ to set up the presets in the first place, which I appreciate is not ideal.

Short of interacting with the GNSS device at the hardware level (i.e., interrogating & programming the MCU and/or EEPROM, which is beyond the scope of a UBX protocol parser), the only way of reproducing the current configuration of a u-blox GNSS device via the UBX protocol is to poll every possible configuration parameter using a series of CFG-* GET queries, and then “replay” these as a series of CFG-* SET commands.

For Generation 9+ devices such as the NEO-M9N, ZED-F9P, etc., legacy CFG-* messages are deprecated in favour of the newer [CFG-VALGET, CFG-VALSET and CFG-VALDEL](https://github.com/semuconsulting/pyubx2#configinterface) message types. You can only poll up to 64 parameters in a single CFG-VALGET message. Given there are over 1,200 possible configuration parameters in the latest [configuration database](https://github.com/semuconsulting/pyubx2/blob/master/pyubx2/ubxtypes_configdb.py) (*not all are applicable to all devices*), a complete configuration dump would involve around 20 individual queries.

Since the UBX protocol does not support explicit synchronous message acknowledgement, there is no absolute guarantee that each CFG poll would receive a timely asynchronous response, especially if the device is already processing high navigation message rates. So you may have to 'retry' some of the polls to ensure you have a complete set of parameters to replay.

But this is not beyond the realms of possibility and I'm happy to put it on the feature backlog for further investigation, but it'll probably be a while before I get to it.

In the short term, I recommend you take a look at PyGPSClient's User-Defined Presets facility and the example `ubxpresets` file to see if it can meet at least some of your needs. If you need any assistance in configuring the `ubxpresets` file, get back to me.

Purely out of interest, are you able to give me a flavour of the sort of configuration(s) you would be looking to apply to multiple u-blox devices? In my experience, most working configurations typically involve changes to no more than 20 default settings, so performing a complete dump of all 1,200 parameters may be overkill.

AgileEngineeringLLC, 2022-12-26:
Thx for the info! It does sound more complex that i anticipated, yikes. I'm doing some work for a client (Axiom Space) where i need to setup F9Ps to the following:L1C only (no other GNSS signals enabled)Log NMEA GGA, NMEA GSA, RXM-RAWX, RXM-MEASXLog rate is NMEA at 250ms, RXM at 1000ms if feasibleI collect mobile data and static data with two F9Ps running concurrently, then process the parsed UBX file data in matlab (and C++) to test out flight nav filter designs. DonDon KellyLinkedIn: www.linkedin.com/in/kellydakCell: 281-221-28534403 Orange Leaf CourtHouston, TX 77059

semuadmin, 2022-12-26:
Hi Don,

Ok so this is a fairly straightforward configuration on the F9P, either using legacy CFG-* commands or the equivalent CFG-VALSET configuration database keys. I’m not sure how familiar you are with the UBX protocol so apologies if I’m stating the obvious here:

Legacy configuration commands:

1)	Set the `measRate` to 250ms using the CFG-RATE command (the default is 1000ms).
2)	Set the message `rate` for the requisite NMEA messages/ports to 1 (that is, one every navigation solution, or every 250ms) using the CFG-MSG command. Set all other NMEA message rates to 0 if you want them suppressed.
3)	Set the message `rate` for the requisite RXM messages/ports to 4 (that is, one every 4th navigation solution, or every 1000ms (250ms * 4)) using the CFG-MSG command.
4)	Filter the requisite GNSS constellation and signal types using the CFG-GNSS command. So if, for example, you wanted only the GPS L1C signal, you would set `gnssId` = 0 (GPS), `enable` = 1, `sigCfgMask` = 0x01. Refer to the UBX F9P Interface Specification for details of other `sigCfgMask` settings.

Equivalent CFG-VALSET configuration database keys (assuming protocol HPG 1.30 of later and device is connected via USB port):

1)	Measurement rate:  `CFG-RATE-MEAS`
2)	Message rate NMEA: `CFG-MSGOUT-NMEA_ID_GGA_USB`, `CFG-MSGOUT-NMEA_ID_GSA_USB` and so on...
3)	Message rate RXM: `CFG-MSGOUT-UBX_RXM_MEASX_USB`, `CFG-MSGOUT-UBX_RXM_RAWX_USB`
4)	GNSS signal filter: `CFG-SIGNAL-GPS_ENA`, `CFG-SIGNAL-GPS_L1CA_ENA`

FYI the [F9P Interface Specification](https://content.u-blox.com/sites/default/files/documents/u-blox-F9-HPG-1.32_InterfaceDescription_UBX-22008968.pdf) has a cross-reference of legacy vs. modern configuration commands in section 6.10.

Hope this helps. If you need pointers setting up these commands as PyGPSClient user-defined preset strings, give me a shout.

AgileEngineeringLLC, 2022-12-26:
Thx, this helps a lot. I had most of the settings, but struggled with a couple. Thx again!

Don

Don Kelly

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

semuadmin, 2023-01-06:
Hi @AgileEngineeringLLC ,

For what it's worth, I've added a couple of command line examples which illustrate how one might save and reload a complete set of configuration parameters from one Generation 9+ GNSS device to anothe (e.g. NEO-M9N or ZED-F9P). 

The save and load processes typically take between 15 and 30 seconds to complete via USB - maybe quicker if you're using a high baud rate UART port.

https://github.com/semuconsulting/PyGPSClient/blob/rc-1.3.14/examples/ubxconfigsave.py
https://github.com/semuconsulting/PyGPSClient/blob/rc-1.3.14/examples/ubxconfigload.py

If these are of any use, I can consider adding this functionality to PyGPSClient in due course.

AgileEngineeringLLC, 2023-01-06:
very nice, thx for the update!DonDon KellyLinkedIn: www.linkedin.com/in/kellydakCell: 281-221-28534403 Orange Leaf CourtHouston, TX 77059

https://github.com/semuconsulting/PyGPSClient/blob/rc-1.3.14/examples/ubxconfigsave.py
https://github.com/semuconsulting/PyGPSClient/blob/rc-1.3.14/examples/ubxconfigload.py

closed 2023-01-12 by semuadmin

AgileEngineeringLLC, 2023-01-23:
Hi,I've started using the ublox SPARTN corrections with a D9S connected to an F9P. The SPARTN keys must be loaded to the F9P by U-Center every time the F9P is powered up. This is done by just pointing U-Center to where you have the text file containing the key code. It'd be terrific if this could be done in pygpdclient!By any chance have you considered adding SPARTN key loading / configuration capability to pygpsclient? Thx!DonDon KellyLinkedIn: www.linkedin.com/in/kellydakCell: 281-221-28534403 Orange Leaf CourtHouston, TX 77059

semuadmin, 2023-01-23:
Don,

Hi again.

If I understand correctly, the SPARTN keys you’re referring to are the decryption keys for an encrypted SPARTN service (e.g. u-blox’s proprietary PointPerfect service). AFAIK they’re not needed for an unencrypted service. See section 3.1.6 of the F9P Integration Manual<https://content.u-blox.com/sites/default/files/ZED-F9P_IntegrationManual_UBX-18010802.pdf>.

I believe they’re sent to the receiver via an RXM-SPARTN-KEY message, so in theory it’s just matter of obtaining the keys, populating a RXM-SPARTN-KEY message and sending it to the receiver in the normal manner.

Two challenges here, one trivial, one not so trivial:

  1.  Firstly, I’ve only just realised I have omitted the RXM-SPARTN-KEY GET/SET message definition from the pyubx2 library, despite it being clearly documented in the F9P HPG 1.3n interface specification. That should be a trivial fix in pyubx2 and I’ll get on that.
  2.  Secondly, the relevant encryption keys are ONLY available via u-blox (or one of their partners) so you’d need to continue to use their service to obtain the keys in the first place, and somehow transfer them over to PyGPSClient under your own steam. There may be some devil in the detail, but AFAICS it should be relatively easy to provide a utility service in PyGPSClient to upload the key and populate a RXM-SPARTN-KEY message automatically. BUT while the SPARTN protocol itself is open source, I’d need to ensure I’m not falling foul of any proprietary u-blox IP.

So the answer is a yes to (1) and a tentative future yes to (2).

semuadmin, 2023-01-25:
Don,

FYI I’ve uploaded an example command line script to the pyubx2 library \examples folder which illustrates how to populate and upload an RXM-SPARTN-KEY message.

https://github.com/semuconsulting/pyubx2/blob/master/examples/spartnkey.py

To run it, simply type python3 spartnkey.py at the command line.

The script prompts you to enter the two SPARTN keys (current and next, normally 16 chars each) and their associated Valid From dates (week number Wno and Time of Week Tow)

Values for Wno and Tow for any given date can be obtained from this GPS Calendar site: https://navigationservices.agi.com/GNSSWeb/
(Wno = “Full GPS week since 1st epoch”; Tow = “seconds of week at midnight for that day”)

You may care to test this out using your DS9 F9P setup. If it works, it would be a relatively simple task for me to incorporate this functionality into PyGPSClient.

Hope this helps

AgileEngineeringLLC, 2023-01-25:
Awesome, thanks!

I may be able to try it out later today, altho weather has not been RTK-friendly the last day or so!

Thx!

Don

Don Kelly

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

https://github.com/semuconsulting/pyubx2/blob/master/examples/spartnkey.py

AgileEngineeringLLC, 2023-01-25:
Hi. I’m getting errors when I try to run the script. Do I need to start up pyGPSClient first? I tried both ways (with and without pyGPSClient connected and running), but am getting errors, so suspect I may be setting things up incorrectly. Here’s an example of what I’m seeing, using the keys that work for me in U-Center:

Last login: Wed Jan 25 14:49:35 on ttys000
***@***.*** ~ % cd desktop
***@***.*** desktop % spartnkey.py
zsh: command not found: spartnkey.py
***@***.*** desktop % python3 spartnkey.py
How many SPARTN keys do you want to upload (1 or 2)? (2): 2
Enter first key as string (max 255 chars): 84e1864a9b587a7a712d88d6e17fe5ce
Enter first valid from Week Number (Wno) as integer: 2248
Enter first valid from Time of Week (Tow) as integer: 518418
Enter second key as string (max 255 chars): 77a3e8cf29c965947164a6fc7740ba9a
Enter second valid from Week Number (Wno) as integer: 2252
Enter second valid from Time of Week (Tow) as integer: 518418
Formatting UBX RXM-SPARTN-KEY message...

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyubx2/ubxmessage.py", line 527, in _get_dict
    pdict = ubs.UBX_PAYLOADS_SET[self.identity]
            ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'RXM-SPARTN-KEY'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/donkelly/Desktop/spartnkey.py", line 67, in <module>
    msg = UBXMessage("RXM", "RXM-SPARTN-KEY", SET, payload=payload)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyubx2/ubxmessage.py", line 81, in __init__
    self._do_attributes(**kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyubx2/ubxmessage.py", line 104, in _do_attributes
    pdict = self._get_dict(**kwargs)  # get appropriate payload dict
            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyubx2/ubxmessage.py", line 554, in _get_dict
    raise KeyError(
KeyError: "'RXM-SPARTN-KEY' - Check 'msgmode' keyword argument is appropriate for message category"
***@***.*** desktop %

semuadmin, 2023-01-25:
you probably need to update your version of pyubx2 to the latest 1.2.21 (I only recently added RCM-SPARTN-KEY support):

python3 -m pip install --upgrade pyubx2

Try this and let me know how you get on

AgileEngineeringLLC, 2023-01-25:
Getting further, but it doesn’t seem to like the port. Also, guessing I should NOT have pygpsclient connected when I run this? Thx!

***@***.*** desktop % python3 spartnkey.py
How many SPARTN keys do you want to upload (1 or 2)? (2): 2
Enter first key as string (max 255 chars): 84e1864a9b587a7a712d88d6e17fe5ce
Enter first valid from Week Number (Wno) as integer: 2248
Enter first valid from Time of Week (Tow) as integer: 518418
Enter second key as string (max 255 chars): 77a3e8cf29c965947164a6fc7740ba9a
Enter second valid from Week Number (Wno) as integer: 2252
Enter second valid from Time of Week (Tow) as integer: 518418
Formatting UBX RXM-SPARTN-KEY message...

<UBX(RXM-SPARTN-KEY, version=1, numKeys=2, reserved0=0, reserved1_01=0, keyLengthBytes_01=32, validFromWno_01=2248, validFromTow_01=518418, reserved1_02=0, keyLengthBytes_02=32, validFromWno_02=2252, validFromTow_02=518418, key_01=56, key_02=52, key_03=101, key_04=49, key_05=56, key_06=54, key_07=52, key_08=97, key_09=57, key_10=98, key_11=53, key_12=56, key_13=55, key_14=97, key_15=55, key_16=97, key_17=55, key_18=49, key_19=50, key_20=100, key_21=56, key_22=56, key_23=100, key_24=54, key_25=101, key_26=49, key_27=55, key_28=102, key_29=101, key_30=53, key_31=99, key_32=101, key_33=55, key_34=55, key_35=97, key_36=51, key_37=101, key_38=56, key_39=99, key_40=102, key_41=50, key_42=57, key_43=99, key_44=57, key_45=54, key_46=53, key_47=57, key_48=52, key_49=55, key_50=49, key_51=54, key_52=52, key_53=97, key_54=54, key_55=102, key_56=99, key_57=55, key_58=55, key_59=52, key_60=48, key_61=98, key_62=97, key_63=57, key_64=97)>

b'\xb5b\x026T\x00\x01\x02\x00\x00\x00 \xc8\x08\x12\xe9\x07\x00\x00 \xcc\x08\x12\xe9\x07\x0084e1864a9b587a7a712d88d6e17fe5ce77a3e8cf29c965947164a6fc7740ba9a\xf9\x88'

Check message formatted correctly: PASS
Do you want to send this message to the receiver? (y/n) y
Enter receiver port: (dev/ttyACM1) dev/cu.usbmodem2101
Enter receiver baud rate: (9600) 9600
Enter receiver timeout: (3) 3
Sending message to ***@***.***

Error sending message! [Errno 2] could not open port dev/cu.usbmodem2101: [Errno 2] No such file or directory: 'dev/cu.usbmodem2101'

To make this message a PyGPSClient user preset, the following 
string can be cut-and-pasted into PyGPSClient's ubxpresets file:

Send RXM-SPARTN-KEY, RXM, RXM-SPARTN-KEY, 010200000020c80812e907000020cc0812e9070038346531383634613962353837613761373132643838643665313766653563653737613365386366323963393635393437313634613666633737343062613961, 1
***@***.*** desktop % 

￼

semuadmin, 2023-01-25:
Like anyting accessing a serial port, the script will require SOLE ACCESS to the serial port so, yes, you'll need to disconnect anything else that's currently accessing the same port, including PyGPSClient. 

Also, are you sure that's the correct serial port designation? Typically you'd want something like "/dev/tty.usbmodem2101" (*tty rather than cu - there is a subtle difference*). But try both.

Ths simple script script runs 'blind', as it were, and you won't immediately see any acknowledgement. 

If you want to see whether the upload was successful, you can poll the RXM-SPARTN-KEY message in PyGPSClient or u-center. Alternatively, the RXM-COR message will report whether you are receiving and decoding corrections from the D9S correctly.

semuadmin, 2023-01-25:
By the way, I'd be careful about broadcasting your SPARTN keys in plain text - Thingstream may not appreciate that!

Please redact them first if you need to send me any further diagnostics.

AgileEngineeringLLC, 2023-01-25:
I added the “/“ in front of dev/cu.modem2101 and it looks like it went (see below). But I’m not seeing the fix change to FLOAT or FIX yet. I also went in and made sure (per the SPARTN guide) that these were set to “1”

CFG-SPARTN-USE_SOURCE
CFG-UART2INPROT-UBX

I also noticed the Lat Lon values are now only changing in the 7 or 8th digit, so it may actually be in FLOAT or FIX but just displaying “fix: 3D”

Don

***@***.*** desktop % python3 spartnkey.py
How many SPARTN keys do you want to upload (1 or 2)? (2): 2
Enter first key as string (max 255 chars): 84e1864a9b587a7a712d88d6e17fe5ce
Enter first valid from Week Number (Wno) as integer: 2248
Enter first valid from Time of Week (Tow) as integer: 518418
Enter second key as string (max 255 chars): 77a3e8cf29c965947164a6fc7740ba9a
Enter second valid from Week Number (Wno) as integer: 2252
Enter second valid from Time of Week (Tow) as integer: 518418
Formatting UBX RXM-SPARTN-KEY message...

<UBX(RXM-SPARTN-KEY, version=1, numKeys=2, reserved0=0, reserved1_01=0, keyLengthBytes_01=32, validFromWno_01=2248, validFromTow_01=518418, reserved1_02=0, keyLengthBytes_02=32, validFromWno_02=2252, validFromTow_02=518418, key_01=56, key_02=52, key_03=101, key_04=49, key_05=56, key_06=54, key_07=52, key_08=97, key_09=57, key_10=98, key_11=53, key_12=56, key_13=55, key_14=97, key_15=55, key_16=97, key_17=55, key_18=49, key_19=50, key_20=100, key_21=56, key_22=56, key_23=100, key_24=54, key_25=101, key_26=49, key_27=55, key_28=102, key_29=101, key_30=53, key_31=99, key_32=101, key_33=55, key_34=55, key_35=97, key_36=51, key_37=101, key_38=56, key_39=99, key_40=102, key_41=50, key_42=57, key_43=99, key_44=57, key_45=54, key_46=53, key_47=57, key_48=52, key_49=55, key_50=49, key_51=54, key_52=52, key_53=97, key_54=54, key_55=102, key_56=99, key_57=55, key_58=55, key_59=52, key_60=48, key_61=98, key_62=97, key_63=57, key_64=97)>

b'\xb5b\x026T\x00\x01\x02\x00\x00\x00 \xc8\x08\x12\xe9\x07\x00\x00 \xcc\x08\x12\xe9\x07\x0084e1864a9b587a7a712d88d6e17fe5ce77a3e8cf29c965947164a6fc7740ba9a\xf9\x88'

Check message formatted correctly: PASS
Do you want to send this message to the receiver? (y/n) y
Enter receiver port: (dev/ttyACM1) /dev/cu.usbmodem2101
Enter receiver baud rate: (9600) 
Enter receiver timeout: (3) 
Sending message to ***@***.***

Message sent

To make this message a PyGPSClient user preset, the following 
string can be cut-and-pasted into PyGPSClient's ubxpresets file:

Send RXM-SPARTN-KEY, RXM, RXM-SPARTN-KEY, 010200000020c80812e907000020cc0812e9070038346531383634613962353837613761373132643838643665313766653563653737613365386366323963393635393437313634613666633737343062613961, 1
***@***.*** desktop %

AgileEngineeringLLC, 2023-01-25:
Will do, thx. Yes, please don’t pass to others, thx!
Don

semuadmin, 2023-01-25:
@AgileEngineeringLLC  OK I've just this minute uploaded an updated version of the `spartnkey.py` script which accepts the key as hexadecimal characters. It also simplifies the entering of the From date - simply enter the date in YYYYMMDD format.

https://github.com/semuconsulting/pyubx2/blob/master/examples/spartnkey.py

Download this updated version and see if you get any joy.

semuadmin, 2023-01-25:
Don - you do realise this conversation is in the public domain, right? :-)

AgileEngineeringLLC, 2023-01-25:
Thanks, I had not realized they were getting posted, so I've deleted certain ones. Thanks.

semuadmin, 2023-01-25:
If you want to contact me directly, use the email address at the foot of the PyGPSClient README.

AgileEngineeringLLC, 2023-01-25:
In pyGPSClient, at the very top right of the GUI should I be seeing"fix:3D" go to "fix:FLOAT" or "fix:FIXED" when we're in RTK lock? I'm wondering what's the easiest means to tell if the solution is actually using SPARTN corrections.

AgileEngineeringLLC, 2023-01-25:
Looks like I can check the GGA messages for the quality flag...

semuadmin, 2023-01-25:
As I mentioned above, you could poll the RXM-COR message, which should tell you what corrections you're receiving. I presume you have good L-band antennae and a goid signal for both the D9S and the F9P? And  I presume you've checked this all works OK using u-center? PS It's very PM on this side of the pond, so I'll leave you to it for now. Can pick up tomorrow if you're still having issues.

AgileEngineeringLLC, 2023-01-25:
Roger, thanks much. I'll keep trying some things. It does seem to be staying in 3D/DGPS mode only, never going to Float or Fixed (per the GGA NMEA message). I have it working well via U-Center, so still likely something I've not enabled correctly.

semuadmin, 2023-01-25:
FYI I've just uploaded the same key values myself into my ZED-F9P using a PyGPSClient user present (which the script generates) and sent an RXM-SPARTN-KEY poll to confirm that they have been uploaded successfully:

[image]

So the keys are definitely being uploaded OK. I don't have a DS9 to hand at the moment so can't test a SPARTN feed.

Just a thought, but the Valid From dates you've entered into the script appear to correspond to dates 11th Feb 2023 (2248 518418) and 11th Mar 2023 (2252 518418), in which case neither of the two keys you're using are actually valid yet. Are those the dates that Thingstream quoted?

AgileEngineeringLLC, 2023-01-25:
Yes, those dates work in U-Center and with an Arduino script I can run (provided by Sparkfun).

semuadmin, 2023-01-25:
ok bizarre - maybe I have an incorrect understanding of 'Valid From' date!

AgileEngineeringLLC, 2023-01-25:
It's an expires date

semuadmin, 2023-01-25:
The ZED-F9P actually needs to know when the keys are valid FROM, rather than when they expire. Each key is valid for four weeks, so we need to work backwards 4 weeks from the expiry date quoted. It's possible that u-center is doing this automatically, so if you enter a date of 11th Feb 2023 it automatically winds this back to 11th Jan 2023. I can easily do the same in the script if this is the case. 

The way to check would be to upload the keys in u-center and then poll for a RXM-SPARTN-KEY message yourself. If the wno and tow are 4 weeks earlier than the date you entered (e.g. 2244 259218 rather than 2248 518418), then we know that's what's happening.

If you want to try this in PyGPSClient, add the following line to the ubxpresets file in your home directory, then restart PyGPSClient:

Poll RXM-SPARTN-KEY, RXM, RXM-SPARTN-KEY, , 2

You can then select this command from the UBX Configuration panel Presets dropdown list. You should see a response similar to the screen shot I posted earlier.

commit 71c8547c (2023-01-10): update config save/load facility  Fixes #36
PR #38 (2023-01-12): Rc 1.3.14
# PyGPSClient Pull Request Template

## Description

ENHANCEMENTS:

1. New CFG Configuration Load/Save/Record facility added to UBX Configuration panel. This allows users to 'record' any configuration commands (UBX CFG messages) sent to a device to an internal memory array, and subsequently save this array in a binary file. Saved files can be reloaded and the messages replayed to any compatible device. This provides a means to easily reproduce a given sequence of configuration commands, or copy a saved configuration between compatible devices.
1. Msg Mode listbox added to serial conf [... cut]