semuconsulting/PyGPSClient #40: Accuracy Estimate Reporting

AgileEngineeringLLC, 2023-01-26:
**Is your feature request related to a problem? Please describe.**
Not a problem, just an idea. It would be really nice to have displayed the current estimate of the filter position accuracy. For example, in U-Center when you're in RTK-Fix you can see that your (reported) accuracy is 0.035m for example, which is very helpful. The reason this is helpful is that RTK-Fix has a fairly wide range of expected accuracy, say 0.015m to more than 0.050m. So this gives you an indication of how well your RTK-Fix lock is that day. Same is true for RTK-Float. [By the way, terrific tool, MUCH easier and more reliable than U-Center!]

A clear and concise description of what the problem is e.g. I'd like to be able to do this [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

**Would you be willing to contribute a test device?**

If the request relates to a specific u-blox device that is not currently supported, would you be 
willing to contribute a device to the project for testing purposes?

**Additional context**

Add any other context about the feature request here.

semuadmin, 2023-01-26:
Hi @AgileEngineeringLLC 

So if I understand correctly, it's the `hAcc` (horizontal accuracy) attribute you're after, in which case this is already supported by PyGPSClient and displayed on the main banner.

However, the default NMEA stream output by u-blox receivers does _not_ contain a `hAcc` value, You have to enable additional message type(s) in order to receive it. Any of the following message types will provide an `hAcc` value:

**NMEA:**

- UBX-00 (a proprietary u-blox NMEA message type only available on u-blox receivers)

**UBX:**

- HNR-PVT
- NAV-HPPOSLLH
- NAV-POSLLH
- NAV-PVAT
- NAV-PVT

If you enable one or more of these message types and are in RTK mode, you should  see the `hAcc` value in PyGPSClient's banner slowly creep down from a new meters to a few centimeters. In the screenshot I display in the PyGPSClient's README, it shows an `hAcc` of 1cm (the minimum reported).

Try this and let me know how you get on.

semuadmin, 2023-01-29:
Hi @AgileEngineeringLLC 

Any joy?

AgileEngineeringLLC, 2023-01-29:
yes, did a test survey a couple of days ago, got horiz accy of about 26mm in rtk fixed, also collected raw ubx measurements at same time. works great!!DonDon KellyLinkedIn: www.linkedin.com/in/kellydakCell: 281-221-28534403 Orange Leaf CourtHouston, TX 77059

semuadmin, 2023-01-29:
Ok I'll close the issue now.

closed 2023-01-29 by semuadmin