semuconsulting/PyGPSClient #48: Saving L-band config into D9S

n8ur, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
There doesn't seem to be a way to save L band configuration info loaded into the NEO-D9S via the SPRTN configuration window.

I was able to set the D9S parameters, but couldn't figure out how to save them to BBRAM/Flash.

**Describe the solution you'd like**

An option to save the D9S configuration in the SPRTN window.

**Describe alternatives you've considered**

I think the only way you could currently save the D9S config is to disconnect from it in the SPRTN window, then reconnect on the main window, go to the UBX config window, and do save config there.  That seems a bit risky, though, as you can't be sure nothing has changed in the connect/reconnect process.

**Would you be willing to contribute a test device?**

I only have one D9S at the moment, but I'm happy to test the code.

semuadmin, 2023-03-09:
It would be easy enough to add a checkbox on the L-Band panel to execute a CFG-CFG save command to BBR / Flash. Would that address the issue?

n8ur, 2023-03-09:
That absolutely would.  Thanks!

semuadmin, 2023-03-09:
> That absolutely would. Thanks!

Done. PR #49 refers.

PR #49 (2023-03-10): Add D9S save config facility Fixes #48
# PyGPSClient Pull Request Template

## Description

Add facility on SPARTN Configuration panel,  L-Band section, to save D9* configuration to persistent storage (BBR / Flash / EEPROM - depending on capabilities of the hardware device).

Fixes #48 

## Testing

Please test all changes, however trivial, against the supplied unittest suite `tests/test_*.py` e.g. by executing the `tests/testsuite.py` module or using your IDE's native Python unittest integration facilities. Please describe any test cases you have amended or added to this suite to maintain >=  [... cut]
commit 01234f6d (2023-03-10): Add D9S save config facility Fixes #48
commit 66d66c4b (2023-03-10): Merge pull request #49 from semuconsulting/add-D9S-config-save

Add D9S save config facility Fixes #48
closed 2023-03-10 by semuadmin