semuconsulting/PyGPSClient #60: UI text unreadable in MacOS with Dark theme

RomanTheLegend, 2023-04-03:
pygpsclient 1.3.20
MacOs Ventura 13.2.1 (22D68)

When Dark theme in MacOS is enabled, the text in inputboxes and lists is unreadable (same FG and BG color)

[image]

This is not the issue with Light theme:
[image]

semuadmin, 2023-04-03:
Hi @RomanTheLegend 

Unfortunately reliable automatic 'dark mode' detection across different O/S platforms is problematic (though not impossible) with tkinter. There are libraries available e.g. [darkdetect](https://pypi.org/project/darkdetect/), but AFAICS they're not yet fully mature.

The easiest way to address this (from an application standpoint) would be to add a user-selectable colormode option to the application itself, via the new json config file facility introduced in v1.3.22, e.g:

{
...
"colormode": "dark"
...
}

The font colours would then be amended accordingly.

Would that be an acceptable solution?

_As an interim workaround, you could amend the following values in globals.py_:

```
BGCOL = "gray24"  # default widget background color
FGCOL = "white"  # default widget foreground color
```

(_As it happens I normally run in 'Dark' mode on Windows and Linux but reverted to 'Light' on my Mac as too many off-the-shelf applications didn't seem to play well with 'Dark' mode._)

RomanTheLegend, 2023-04-03:
Yep, that's what I was looking for, thanks

semuadmin, 2023-04-03:
> Yep, that's what I was looking for, thanks

OK I'll add this to v1.3.23 - may be a couple of weeks.

PR #61 (2023-04-15): Rc 1.3.23
# PyGPSClient Pull Request Template

## Description

RELEASE CANDIDATE v1.3.23

CHANGES:

1. Improved legibility in 'Light' and 'Dark' O/S modes by removing hard-coded Entry widget background colors.
1. Minimum pygnssutils version updated to 1.0.6.
1. Minor fixes to configuration load status messages.

Fixes #60 

## Testing

Please test all changes, however trivial, against the supplied pytest suite `tests/test_*.py`. Please describe any test cases you have amended or added to this suite to maintain >= 99% code coverage.

- [x] static tests updated

## Checklist:

- [x] I agree to a [... cut]
closed 2023-04-15 by semuadmin