semuconsulting/PyGPSClient #84: Raspberry Pi 32-bit

acbtones, 2023-08-04:
### Discussed in https://github.com/semuconsulting/PyGPSClient/discussions/83

<div type='discussions-op-text'>

<sup>Originally posted by **acbtones** August  4, 2023</sup>
Hi,

Firstly, thanks for the great software!
However, I can't get it to install on Raspberry Pi 32-bit OS.
Using a RPi 3, I installed 32bit OS ("Raspberry Pi OS Full (32-bit)") and tried the installation - log attached: 32bit_pygpsclient.log
Cryptography fails to install so the PyGPSClient install fails.
Is that related to this: https://github.com/pyca/cryptography/issues/5766#issuecomment-778538695
I think that's why previous discussions in this forum have suggested Rust is required.
Using the same RPi 3, I installed 64-bit OS ("Raspberry Pi OS (64-bit)") and tried the installation - log attached: 64bit_pygpsclient.log
Successful installation.

My wish is to add PyGPSClient to an existing 32-bit RPi installation running other software that will only run on 32-bit.

Also, when starting PyGPSClient, is it possible to start the stream immediately i.e. without clicking the USB button?

Thanks,

Anthony

[32bit_pygpsclient.log](https://github.com/semuconsulting/PyGPSClient/files/12256451/32bit_pygpsclient.log)
[64bit_pygpsclient.log](https://github.com/semuconsulting/PyGPSClient/files/12256452/64bit_pygpsclient.log)

</div>

semuadmin, 2023-08-04:
Hi @acbtones 

Sorry you're having difficulties.

PyGPSClient uses the pyspartn library, which in turn uses the cryptography library, for decrypting SPARTN messages.

It appears that from cryptography >=38.0 onwards,  if you install cryptography via pip from a source file (.tar.gz) rather than a binary wheel (.whl), the installation requires a rust compiler. Most modern 64-bit Linux distros can provide a wheel distribution package, but it appears that some older 32-bit versions of Raspi OS don't.

_NB: this is only an issue if pip needs to install cryptography from source_ - if pip can find a binary wheel installation package for your platform, the problem should not occur.

Sometimes simply updating pip can resolve the problem (_by making available a wheel distribution package_)...

`python3 -m pip install --upgrade pip`

... though I tried this on my own RPI 3 running Raspi Lite without success.

The following worked for me on an RPI 3 running 32-bit Raspi OS Lite:

1. Firstly, install the rust compiler dependencies:

   `sudo apt-get install build-essential libssl-dev libffi-dev python3-dev cargo pkg-config`

2. Then install cryptography (**NB: the compilation will take several minutes on an old 32-bit RPi**):

   `python -m pip install cryptography`

```
pi@raspberrypi:~ $ python -m pip install cryptography
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cryptography
  Using cached cryptography-41.0.3.tar.gz (630 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting cffi>=1.12 (from cryptography)
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting pycparser (from cffi>=1.12->cryptography)
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml) ... | 
done
  Created wheel for cryptography: filename=cryptography-41.0.3-cp39-cp39-linux_armv7l.whl size=2084218 sha256=6ddc66713c9cd8a74db131edeb1519a34cbb5e34d1d85f0db7c4662efb9a9aa1
  Stored in directory: /home/pi/.cache/pip/wheels/83/ef/54/3d54fa97a47ec9b505300e38f091780aadce82b0d74f5bcd1c
Successfully built cryptography
Installing collected packages: pycparser, cffi, cryptography
Successfully installed cffi-1.15.1 cryptography-41.0.3 pycparser-2.21
```

3. You should then be able to install pyspartn (and by implication any packages dependent on it):

  `python3 -m pip install pyspartn`

```
pi@raspberrypi:~ $ python3 -m pip install pyspartn
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyspartn
  Downloading https://www.piwheels.org/simple/pyspartn/pyspartn-0.1.10-py3-none-any.whl (26 kB)
Requirement already satisfied: cryptography>=39.0.2 in ./.local/lib/python3.9/site-packages (from pyspartn) (41.0.3)
Requirement already satisfied: cffi>=1.12 in ./.local/lib/python3.9/site-packages (from cryptography>=39.0.2->pyspartn) (1.15.1)
Requirement already satisfied: pycparser in ./.local/lib/python3.9/site-packages (from cffi>=1.12->cryptography>=39.0.2->pyspartn) (2.21)
Installing collected packages: pyspartn
Successfully installed pyspartn-0.1.10
```

Try this and let me know how you get on.

If it works, I'll add these instructions to the README.md for PYGPSClient and pyspartn.

semuadmin, 2023-08-04:
@acbtones 

In answer to your second point about automatically starting a given serial connection when PyGPSClient starts, this is not supported, but what you _can_ do is start PyGPSClient with a comand line option -U or --userport, e.g.:

`pygpsclient --userport /dev/ttyACM1`

The designated serial port will then be automatically pre-selected in the serial port selection listbox, but you will still need to press the serial 'connect' button.

semuadmin, 2023-08-04:
@acbtones 

As an aside, I suspect relatively few PyGPSClient users are actually interested in decrypting SPARTN messages (_and the pyspartn library is still only in Alpha_), so I may look to add an environment variable e.g. `PYSPARTN_NODECRYPT` which in effect suppresses the dependency on the cryptography library for those users who are happy to live without this functionality.

But, hopefully, the wheel distribution package for cryptography will eventually find its way into older 32-bit distros.

acbtones, 2023-08-04:
>    1. You should then be able to install pyspartn (and by implication any

Hi,

Thanks for responding so quickly and thanks for the suggestions, but it
still doesn't work.
From a standard Raspberry Pi OS Full, 32-bit, I'm only doing updates and
then trying your suggestions:
sudo apt-get update
sudo apt-get upgrade
sudo reboot
python3 -m pip install --upgrade pip
sudo apt-get install build-essential libssl-dev libffi-dev python3-dev
cargo pkg-config
python3 -m pip install cryptography

*Defaulting to user installation because normal site-packages is not
writeableLooking in indexes: https://pypi.org/simple
<https://pypi.org/simple>, https://www.piwheels.org/simple
<https://www.piwheels.org/simple>Requirement already satisfied:
cryptography in /usr/lib/python3/dist-packages (3.3.2)*

python3 -m pip install pyspartn - failed: log attached.

Regards,

Anthony

python3 -m pip install pyspartn

Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyspartn
  Downloading https://www.piwheels.org/simple/pyspartn/pyspartn-0.1.10-py3-none-any.whl (26 kB)
Collecting cryptography>=39.0.2 (from pyspartn)
  Downloading cryptography-41.0.3.tar.gz (630 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 630.1/630.1 kB 2.2 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting cffi>=1.12 (from cryptography>=39.0.2->pyspartn)
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting pycparser (from cffi>=1.12->cryptography>=39.0.2->pyspartn)
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml): started
  Building wheel for cryptography (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error

  × Building wheel for cryptography (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [160 lines of output]
      /tmp/pip-build-env-7s_u2bya/overlay/lib/python3.9/site-packages/setuptools/config/pyprojecttoml.py:66: _BetaConfiguration: Support for `[tool.setuptools]` in `pyproject.toml` is still *beta*.
        config = read_configuration(filepath, True, ignore_option_errors, dist)
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-39
      creating build/lib.linux-aarch64-cpython-39/cryptography
      copying src/cryptography/__about__.py -> build/lib.linux-aarch64-cpython-39/cryptography
      copying src/cryptography/fernet.py -> build/lib.linux-aarch64-cpython-39/cryptography
      copying src/cryptography/utils.py -> build/lib.linux-aarch64-cpython-39/cryptography
      copying src/cryptography/__init__.py -> build/lib.linux-aarch64-cpython-39/cryptography
      copying src/cryptography/exceptions.py -> build/lib.linux-aarch64-cpython-39/cryptography
      creating build/lib.linux-aarch64-cpython-39/cryptography/hazmat
      copying src/cryptography/hazmat/_oid.py -> build/lib.linux-aarch64-cpython-39/cryptography/hazmat
      copying src/cryptography/hazmat/__init__.py -> build/lib.linux-aarch64-cpython-39/cryptography/hazmat
      creating build/lib.linux-aarch64-cpython-39/cryptography/x509
[... 124 lines cut ...]
          successfully install cryptography:
          1) Upgrade to the latest pip and try again. This will fix errors for most
             users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
          2) Read https://cryptography.io/en/latest/installation/ for specific
             instructions for your platform.
          3) Check our frequently asked questions for more information:
             https://cryptography.io/en/latest/faq/
          4) Ensure you have a recent Rust toolchain installed:
             https://cryptography.io/en/latest/installation/#rust

          Python: 3.9.2
          platform: Linux-6.1.21-v8+-aarch64-with-glibc2.31
          pip: n/a
          setuptools: 68.0.0
          setuptools_rust: 1.6.0
          rustc: 1.48.0 (471d696c7 2020-12-10)
          =============================DEBUG ASSISTANCE=============================

      error: Rust 1.48.0 does not match extension requirement >=1.56.0
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects

semuadmin, 2023-08-05:
Hi @acbtones 

OK so it appears to be objecting to the specific version of the rust compiler installed. Let me install everything from scratch so we're comparing like with like...

1. I installed a fresh Raspberry Pi OS 32-bit Full on my RPI3 Model B.2 using the official Raspberry Pi Imager, so it should exactly match your machine. I did an initial update:

```shell
sudo apt update
sudo apt upgrade -y
sudo apt autoremove -y
sudo reboot
```

The platfom details after rebooting were as follows:

```shell
cat /etc/os-release
```
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

```shell
uname -a
```
```
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
```

```shell
python3 -V
```
```
Python 3.9.2
```

2. Install Rust using the official instructions [here](https://www.rust-lang.org/tools/install). This takes about 3 minutes to run:

```shell
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
```

3.  Check the version of the rust compiler installed (it needs to be >= 1.56 apparently):

```shell
rustc -V
```
```
rustc 1.71.1 (eb26296b5 2023-08-03)
````

4. Install the cryptography library build dependencies (omitting `cargo`, which has already been installed in the previous step):

```shell
sudo apt-get install build-essential libssl-dev libffi-dev python3-dev pkg-config
```

5. Update pip:

```shell
python3 -m pip install --upgrade pip
```

6. Install pyspartn using pip (this will take about 2 minutes as it has to compile the `cryptography` library from a source `.tar.gz` file):

```shell
python3 -m pip install pyspartn
```
```
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyspartn
  Using cached https://www.piwheels.org/simple/pyspartn/pyspartn-0.1.10-py3-none-any.whl (26 kB)
Collecting cryptography>=39.0.2 (from pyspartn)
  Using cached cryptography-41.0.3.tar.gz (630 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting cffi>=1.12 (from cryptography>=39.0.2->pyspartn)
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting pycparser (from cffi>=1.12->cryptography>=39.0.2->pyspartn)
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml) ... done
  Created wheel for cryptography: filename=cryptography-41.0.3-cp39-cp39-linux_armv7l.whl size=2084768 sha256=9aa029edb05e0337f363133d1b58b7be518a4d79e905a4361994323f766ded0b
  Stored in directory: /home/steve/.cache/pip/wheels/83/ef/54/3d54fa97a47ec9b505300e38f091780aadce82b0d74f5bcd1c
Successfully built cryptography
Installing collected packages: pycparser, cffi, cryptography, pyspartn
Successfully installed cffi-1.15.1 cryptography-41.0.3 pycparser-2.21 pyspartn-0.1.10
```

7. If this is successful, you should then be able to install PyGPSClient as per the standard installation instructions, remembering to install the `python3-pil.imagetk` dependency first (*it's not installed by default on Raspberry Pi OS*):

```shell
sudo apt install python3-pil.imagetk
```
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  python-pil-doc python3-pil.imagetk-dbg
The following NEW packages will be installed:
  python3-pil.imagetk
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 74.4 kB of archives.
After this operation, 103 kB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf python3-pil.imagetk armhf 8.1.2+dfsg-0.3+deb11u1 [74.4 kB]
Fetched 74.4 kB in 1s (129 kB/s)             
Selecting previously unselected package python3-pil.imagetk:armhf.
(Reading database ... 106702 files and directories currently installed.)
Preparing to unpack .../python3-pil.imagetk_8.1.2+dfsg-0.3+deb11u1_armhf.deb ...
Unpacking python3-pil.imagetk:armhf (8.1.2+dfsg-0.3+deb11u1) ...
Setting up python3-pil.imagetk:armhf (8.1.2+dfsg-0.3+deb11u1) ...
```

```shell
python3 -m pip install pygpsclient
```
```
...
Successfully installed Pillow-10.0.0 charset-normalizer-3.2.0 paho-mqtt-1.6.1 pygnssutils-1.0.12 pygpsclient-1.4.0 pynmeagps-1.0.28 pyrtcm-1.0.11 pyserial-3.5 pyubx2-1.2.29 requests-2.31.0
```

And here is PyGPSClient running quite happily on my RPI3 32-bit Full machine:

[image]

Hope this helps.

semuadmin, 2023-08-05:
@acbtones

If all else fails, here's a zipped copy of the actual `*.whl` file generated when I installed `cryptography` on my RPI3 above. 

**NB:** 
1) This will only work on RPI3 Model B platforms.
2) Downloading installation files from untrusted sources is not normally to be recommended, so install at your own risk. Check the hash on the unzipped file to ensure it hasn't been tampered with before installing:

```shell
shasum -a 256 cryptography-41.0.3-cp39-cp39-linux_armv7l.whl

9aa029edb05e0337f363133d1b58b7be518a4d79e905a4361994323f766ded0b  cryptography-41.0.3-cp39-cp39-linux_armv7l.whl
```

Once unzipped, you should be able to install this directly using:

```shell
python3 -m pip install cryptography-41.0.3-cp39-cp39-linux_armv7l.whl
```

[cryptography-41.0.3-cp39-cp39-linux_armv7l.whl.zip](https://github.com/semuconsulting/PyGPSClient/files/12267014/cryptography-41.0.3-cp39-cp39-linux_armv7l.whl.zip)

acbtones, 2023-08-07:
>    1. Install Rust using the official instructions here
>    1. Check the version of the rust compiler installed (it needs to be >=
>    1. Install the cryptography library build dependencies (omitting cargo,
>    1. Update pip:
>    1. Install pyspartn using pip (this will take about 2 minutes as it
>    1. If this is successful, you should then be able to install

Hi,

Thanks again for your comprehensive follow up!
I can confirm that your updated instructions work on a RPi3, 32-bit.
I also tried the installation on a RPi4, 32-bit and there is one catch that
users should be aware of: when installing Rust on the RPi4, 32-bit, the
default Rust installation is:

Current installation options:
   default host triple: aarch64-unknown-linux-gnu
     default toolchain: stable (default)
               profile: default
  modify PATH variable: yes

However, the default host triple: aarch64-unknown-linux-gnu is for a 64-bit
OS.
The Rust installer is not detecting that the RPi is running the 32-bit OS.
The installation appears to succeed, but Rust is not installed:

info: default toolchain set to 'stable-aarch64-unknown-linux-gnu'

  stable-aarch64-unknown-linux-gnu installed - (error reading rustc version)

Rust is installed now. Great!

To get started you may need to restart your current shell.
This would reload your PATH environment variable to include
Cargo's bin directory ($HOME/.cargo/bin).

To configure your current shell, run:
source "$HOME/.cargo/env"
$ source "$HOME/.cargo/env"
$ rustc -V
error: command failed: 'rustc': No such file or directory (os error 2)

So you have to customise the Rust installation (Option 2) and use
the armv7-unknown-linux-gnueabihf host triple:

Current installation options:

   default host triple: aarch64-unknown-linux-gnu
     default toolchain: stable (default)
               profile: default
  modify PATH variable: yes

1) Proceed with installation (default)
2) Customize installation
3) Cancel installation
>2

I'm going to ask you the value of each of these installation options.
You may simply press the Enter key to leave unchanged.

Default host triple? [aarch64-unknown-linux-gnu]
armv7-unknown-linux-gnueabihf

Default toolchain? (stable/beta/nightly/none) [stable]

Profile (which tools and data to install)? (minimal/default/complete)
[default]

Modify PATH variable? (Y/n)
y

Current installation options:

   default host triple: armv7-unknown-linux-gnueabihf
     default toolchain: stable
               profile: default
  modify PATH variable: yes

1) Proceed with installation (default)
2) Customize installation
3) Cancel installation
>1

Rust installs successfully.
$ source "$HOME/.cargo/env"
$ rustc -V
rustc 1.71.1 (eb26296b5 2023-08-03)

RPi3:

[image: RPi3_32bit.png]

RPi4:

[image: RPi4_32bit.png]

Regards,

Anthony

semuadmin, 2023-08-07:
> Hi, Thanks again for your comprehensive follow up! I can confirm that your updated instructions work on a RPi3, 32-bit. I also tried the installation on a RPi4, 32-bit and there is one catch that users should be aware of: when installing Rust on the RPi4, 32-bit...

Thanks for the tip - just confirmed this on my own RPI 4 Model B 4MB. I've come across this issue before - the platform architecture of the RPI4 is indeed `aarch64`, but when running 32-bit OS on this platform, pip and other package managers do not recognise this as a compatible 'tag' for installation files, so you have to substitute `armv7l`:

```shell
uname -a

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

```shell
python3 -m pip debug --verbose
```
```
...
Compatible tags: 369
...
  cp39-cp39-linux_armv7l
  cp39-abi3-manylinux_2_31_armv7l
  cp39-abi3-manylinux_2_30_armv7l
...
```

Again, for those who would prefer to avoid the palaver of installing Rust and all its dependencies, here's a zipped `.whl` file for `cryptography 41.0.3` for the RPI4 platform, to use at your own risk (once again, **check the hash matches** once you've unzipped and copied this to your RPI):

[cryptography-41.0.3-cp39-cp39-linux_armv7l.whl.zip](https://github.com/semuconsulting/PyGPSClient/files/12273663/cryptography-41.0.3-cp39-cp39-linux_armv7l.whl.zip)

```shell
shasum -a 256 cryptography-41.0.3-cp39-cp39-linux_armv7l.whl
```
```
98fa0f713078d2036a7bb83c9f37bb7c3446ab7f2367d4b2881b259b6baca9fd  cryptography-41.0.3-cp39-cp39-linux_armv7l.whl
```
```shell
python3 -m pip install cryptography-41.0.3-cp39-cp39-linux_armv7l.whl
```
```
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./cryptography-41.0.3-cp39-cp39-linux_armv7l.whl
Requirement already satisfied: cffi>=1.12 in /home/steve/.local/lib/python3.9/site-packages (from cryptography==41.0.3) (1.15.1)
Requirement already satisfied: pycparser in /home/steve/.local/lib/python3.9/site-packages (from cffi>=1.12->cryptography==41.0.3) (2.21)
Installing collected packages: cryptography
Successfully installed cryptography-41.0.3
```

Oh, and thanks for the donation - very much appreciated 👍

semuadmin, 2023-08-07:
@acbtones

I've updated the installation documentation in PyGPSClient and pyspartn.

Are you happy to close this issue as resolved now?

acbtones, 2023-08-07:
Yes, all good and once again thanks for your help.

closed 2023-08-07 by semuadmin