semuconsulting/PyGPSClient #86: incomplete configuration saving and loading

asherikov, 2023-08-07:
I am having weird issues with configuration files using pygpsclient version 1.4.0, installed with pip on Ubuntu 22.04:
- serial port & ntrip client configurations are not saved: ntrip parameters are always set to default, I am not sure if serial port configuration is saved at all;
- if I try to change ntrip parameters manually and then load them, they are set to defaults;
- socket server port is saved to configuration file if I change it in GUI, but is not loaded.

semuadmin, 2023-08-07:
Hi @asherikov 

A fix for these issues (and some other config-related 'quirks') is already in hand and should be available in RC 1.4.1 in the next week or so.

Serial port configurations are not currently stored as there's no way of knowing which serial devices will be connected from one session to the next, but you can set a preferred 'userport' value (either by manually editing the "userport" value in the json config file, setting the `--userport` command line argument on startup, or via environment variable PYGPSCLIENT_USERPORT). If set, this will be pre-selected in the serial port listbox, regardless of whether the device actually exists.

Sorry for any inconvenience.

asherikov, 2023-08-07:
>    A fix for these issues (and some other config-related 'quirks') is
>    already in hand and should be available in RC 1.4.1 in the next week or
>    so.
Thanks for update.

>    Serial port configurations are not currently stored as there's no way
>    of knowing which serial devices will be connected from one session to
>    the next,
That would actually be another feature request: I create symlinks for
devices based on their hardware ids using udev rules to address this issue.
It would be nice to have an option to specify those symlinks manually, e.g.,
`/dev/my_gps_device`.

--
With best regards, A. Sherikov.

asherikov, 2023-08-08:
Ok, thanks, I'll give it a shot when new version is released.

closed 2023-08-08 by asherikov