semuconsulting/PyGPSClient #88: MacOSX 11.7.9 no idea what's wrong

MW0SBX, 2023-08-29:
marc@marcs-MacBook-Pro ~ % python3 -m pygpsclient
/Users/marc/Library/Python/3.8/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
macOS 11 (1107) or later required, have instead 11 (1106) !
zsh: abort      python3 -m pygpsclient
marc@marcs-MacBook-Pro ~ %

semuadmin, 2023-08-29:
Hi @MW0SBX 

Sorry you're having difficulties, but - failing Google - I'm afraid you may need to direct this question to the maintainer of the [urllib3 library](https://github.com/urllib3/urllib3/issues), which is the one that seems to be giving you grief. PyGPClient itself has no pinned OpenSSL version dependency.

If you Google the error, there appear to be quite a few suggested resolutions e.g. https://stackoverflow.com/questions/76187256/importerror-urllib3-v2-0-only-supports-openssl-1-1-1-currently-the-ssl-modu. The options appear to come down to:
1. Install an older version of urllib3
2. Install a newer version of OpenSSL

**Note, however, that**:
1. You may need to use homebrew or macports to do this if pip can't resolve from standard MacOS repos. 
2. You may need to install PyGPSClient in its own virtual environment to avoid version conflicts with any other Python apps you may have (see README for instructions).
3. Running older versions of cryptographic libraries like OpenSSL may leave your machine vulnerable to cyber attacks, though I'm guessing your Mac is > ≈8 years old if it won't upgrade beyond 11.7.9 (Big Sur), so that ship may have sailed?

Unfortunately I no longer have a machine that old to play with (_mainly for professional security compliance reasons_).

MW0SBX, 2023-08-29:
Maybe you need to stop using urllib3 because it stops this project working on any Mac I used opencore patcher and upgraded to osx 12.6.8 and it still complains about urllib3 and still doesn't work.

semuadmin, 2023-08-29:
> Maybe you need to stop using urllib3 because it stops this project working on any Mac I used opencore patcher and upgraded to osx 12.6.8 and it still complains about urllib3 and still doesn't work.

I can assure you that PyGPSClient works fine on MacOS 12 or 13 using the official [Python.org](https://www.python.org/) version of Python (_I'm using it on MacOS 13.5.1 as we speak - it's my standard development environment_), and many others have installed it without difficulty on older MacOS 11 machines. I believe the problem is not with PyGPSClient or urllib3 - it's with the specific version of OpenSSL you have installed on your machine, as the error message tells you.

PyGPSClient doesn't actually use `urllib3` explicitly - it uses the standard `requests` library to retrieve any internet resources it needs, but this in turn has a lower-level dependency on `urllib3` which I have absolutely no control over. I'm not about to write a generic Python internet protocol handler from scratch.

Have you tried installing PyGPSClient using virtualenv? This should in theory provide an 'insulated environment' into which pip can install all the relevant library dependencies without coming into conflict with external apps.

MW0SBX, 2023-08-29:
At the moment it complaining about Tkinter

MW0SBX, 2023-08-29:
i tried sudo apt install tk-dev it then complains about java i went and got java and it still complains that its not installed.

semuadmin, 2023-08-29:
By way of illustration, I just tried a fresh installation on the oldest MacBook I have to hand - an early 2015 MacBook Air running Monterey 12.6.8 on Intel i5. It works fine - see screen shot below...
[image]

MW0SBX, 2023-08-29:
Doesn't work here and i have been at this hours.

[Screenshot 2023-08-29 at 22 42 14]<https://user-images.githubusercontent.com/28569967/264156096-4a864a04-9c9b-42ad-84e2-48095ec4726a.png>

semuadmin, 2023-08-29:
> i tried sudo apt install tk-dev it then complains about java i went and got java and it still complains that its not installed. Sent from Outlook for Android<https://aka.ms/AAb9ysg>
> […](#) [...]

[image]

As explained in the README.md, the version of Python that is installed by default on even the latest MacBooks contains a long-deprecated version of tkinter - 8.5). To get round this, you simply need to install **the latest official version of Python from Python.org**. This will install tkinter 8.6 along with all the other necessary dependencies. I've literally just done this on my old 2015 MacBook Air - it took less than 2 minutes and, as you can see above, it works fine.

MW0SBX, 2023-08-29:
i did i went to python site and installed it ages ago

[Screenshot 2023-08-29 at 21 50 01]<https://user-images.githubusercontent.com/28569967/264157520-58baf938-8328-493a-978a-fd8054940d4e.png>

MW0SBX, 2023-08-29:
python 3.11.5 is install and it still says Tk isnt there

semuadmin, 2023-08-29:
@MW0SBX 

I am genuinely sorry you're having a hard time installing on your machine, but short of reinstalling MacOS from scratch (_from the official Apple source - not an opencore patch_) and then installing a fresh copy of Python 3.11 from Python.org, I'm not sure what more I can suggest.

As I say, PyGPSClient has been installed on thousands of MacOS machines - both new and old - without difficulty _provided_ the user has an official version of Python >=3.8 with tkinter >=8.6 installed. I've no idea why Apple still bundles tkinter 8.5 - it's been deprecated for well over 10 years - but I have no control over that.

Would you be able to send me the complete log and traceback from the latest attempted PyGPSClient installation - the one that is now complaining about Tk? i.e. run this command and send me the complete `log.txt` file:

```shell
python3 -m pip install PyGPSClient > log.txt
```

Can you confirm what you get in response to the following commands (_I've illustrated the responses I'm getting on my machine by way of comparison_):

from the terminal:

```shell
which python3
python3 -V
```

/Library/Frameworks/Python.framework/Versions/3.11/bin/Python3
Python 3.11.4

from an interactive Python3 prompt:

```python
import tkinter
print(tkinter.TclVersion)
tcl = tkinter.Tcl().
print(tcl.call("info", "patchlevel"))
```

8.6
'8.6.12'

MW0SBX, 2023-08-29:
Requirement already satisfied: PyGPSClient in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (1.4.2)
Requirement already satisfied: requests>=2.28.0 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from PyGPSClient) (2.31.0)
Requirement already satisfied: Pillow>=9.0.0 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from PyGPSClient) (10.0.0)
Requirement already satisfied: pygnssutils>=1.0.13 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from PyGPSClient) (1.0.14)
Requirement already satisfied: pyserial>=3.5 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from PyGPSClient) (3.5)
Requirement already satisfied: pyspartn>=0.1.10 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from PyGPSClient) (0.1.10)
Requirement already satisfied: paho-mqtt>=1.6.1 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from pygnssutils>=1.0.13->PyGPSClient) (1.6.1)
Requirement already satisfied: pyubx2>=1.2.29 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from pygnssutils>=1.0.13->PyGPSClient) (1.2.29)
Requirement already satisfied: cryptography>=39.0.2 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from pyspartn>=0.1.10->PyGPSClient) (41.0.3)
Requirement already satisfied: charset-normalizer<4,>=2 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from requests>=2.28.0->PyGPSClient) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from requests>=2.28.0->PyGPSClient) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from requests>=2.28.0->PyGPSClient) (2.0.4)
Requirement already satisfied: certifi>=2017.4.17 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from requests>=2.28.0->PyGPSClient) (2023.7.22)
Requirement already satisfied: cffi>=1.12 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from cryptography>=39.0.2->pyspartn>=0.1.10->PyGPSClient) (1.15.1)
Requirement already satisfied: pynmeagps>=1.0.28 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from pyubx2>=1.2.29->pygnssutils>=1.0.13->PyGPSClient) (1.0.28)
Requirement already satisfied: pyrtcm>=1.0.11 in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from pyubx2>=1.2.29->pygnssutils>=1.0.13->PyGPSClient) (1.0.12)
Requirement already satisfied: pycparser in ***@***.***/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages (from cffi>=1.12->cryptography>=39.0.2->pyspartn>=0.1.10->PyGPSClient) (2.21)

semuadmin, 2023-08-29:
I'm only seeing a partial installation log there (which looks fine) - do you have the complete log?

MW0SBX, 2023-08-29:
how do i do that

semuadmin, 2023-08-29:
if you run the command exactly as follows...  

```shell
python3 -m pip install PyGPSClient > log.txt
```

... you should end up with a file named `log.txt` in the same directory. Just send me the complete file - if it's too big to upload to GitHub, send it as an email attachment to semuadmin@semuconsulting.com.

MW0SBX, 2023-09-11:
> Hi @MW0SBX<https://github.com/MW0SBX>
>
> Are you able to provide the installation log requested?
>
> Happy to continue to investigate, but I'll close the issue in 5 days if no further response.

I got it working using i think a cleanup using Brew a few days ago.

semuadmin, 2023-09-11:
> I got it working using i think a cleanup using Brew a few days ago. 

Ah OK great. I'll close this now.

closed 2023-09-11 by semuadmin