semuconsulting/PyGPSClient #101: Pygpsclient crashes when "save configuration"  on macos

jxl77, 2023-12-28:
# PyGPSClient Bug Report Template

**Describe the bug**
I tried several times, 
but as soon as I  ask to save configuration, PYGPSClient crashes immediately 

I was connected with serial port  with the following configuration 

Process:               python3.10 [23489]
Path:                  /Users/USER/*/python
Identifier:            python3.10
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [4478]
Responsible:           Terminal [4934]
User ID:               502
Date/Time:             2023-12-28 21:59:02.3466 +0100
OS Version:            macOS 12.7 (21G816)
Model: MacBookPro18,2, BootROM 10151.1.1, proc 10:8:2 processors, 64 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: DELL UP3216Q, 6016 x 3384, Main, MirrorOff, Online
Display: Color LCD, 3456 x 2234 Retina, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Apr  6 2022 05:55:54 version 20.90.45.0.8.7.118 FWID 01-e7138ff2
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB3.1 Hub
USB Device: AX88179A
USB Device: USB2.0 Hub
USB Device: USB2.1 Hub
USB Device: USB 2.0 Hub
USB Device: FT232R USB UART
USB Device: USB Mouse
USB Device: USB Billboard Device
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: WD19TB Thunderbolt Dock, Dell, 1, 60.60

**pygpsclient 1.4.6**

Translated Report (Full Report Below)
-------------------------------------

Process:               python3.10 [23489]
Path:                  /Users/USER/*/python
Identifier:            python3.10
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [4478]
Responsible:           Terminal [4934]
User ID:               502

Date/Time:             2023-12-28 21:59:02.3466 +0100
OS Version:            macOS 12.7 (21G816)
Report Version:        12
Anonymous UUID:        ED00EC5F-84FF-208C-8B11-D7D296943ACF

Sleep/Wake UUID:       5E1BCE4F-9D11-41EE-A854-528A5218764D

Time Awake Since Boot: 2000000 seconds
Time Since Wake:       547106 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread
[... 521 lines cut ...]

Model: MacBookPro18,2, BootROM 10151.1.1, proc 10:8:2 processors, 64 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: DELL UP3216Q, 6016 x 3384, Main, MirrorOff, Online
Display: Color LCD, 3456 x 2234 Retina, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Apr  6 2022 05:55:54 version 20.90.45.0.8.7.118 FWID 01-e7138ff2
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB3.1 Hub
USB Device: AX88179A
USB Device: USB2.0 Hub
USB Device: USB2.1 Hub
USB Device: USB 2.0 Hub
USB Device: FT232R USB UART
USB Device: USB Mouse
USB Device: USB Billboard Device
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: WD19TB Thunderbolt Dock, Dell, 1, 60.60

semuadmin, 2023-12-28:
Hi @jxl77, 

Sorry you're having difficulties. The 'Save Configuration' facility has been tested on a variety of Apple MacOS devices - both Intel and Apple Silicon - and I've not come across this issue before. I'm unable to reproduce the error on my own 2021  Macbook Pro (Apple M1 Max) running MacOS Sonama 14.2.1 and using versions of Python from 3.8 to 3.12.

When you say PyGPSClient 'crashes', is there a Python error traceback available at the terminal? Unfortunately the detailed configuration data doesn't really help me much.

Are you running PyGPSClient in a virtual environment or VM?

Where are you trying to save the configuration to?

Other than this error, does PyGPSClient run normally?

I note you're running MacOS 12 (Monterey) - do I take it your machine doesn't support later versions? I also note you're using one or two external monitors - does the issue still occur if you disconnect those monitors and just use the Macbook's native screen?

jxl77, 2023-12-29:
Thanks for this rapid answer.

I am running PyGPSClient   using the PyGPSClient  command line in a terminal.
I have closed all applications, rebooted the computer,  suppress the external monitor,
started again, but with the same result  :

I   connect to the gps through USB/serial, at  115200 bds,  properly receive the flux,
then go to file/save configuration,

and it crashes.

All the other items seem to work properly. I have changed the baud rate of the gps using the UBX commands and it was properly changed in the GPS.
try several other items,
Only the save/configuration has a problem, either by the menu or by the icon in the UBX panel.
I am using Monterey because  I dont rush  to new systems because of uncertainty of behaviour of  many applications I use. I prefer a conservative approach,
than spending my time in upgrades.

Thanks for any help

Jacques

semuadmin, 2023-12-29:
1. Assuming you're running PyGPSClient from the command line: 

```shell
pygpsclient
```

... could you please forward the **full error traceback log** from the terminal after the 'crash'?

2. Could you confirm how you installed the Python 3.10 environment on this machine? Was it via an official [Python.org](https://www.python.org/downloads/) download?

3. Could you also try to record a datalog and/or GPX track to the same location that you are attempting to store the configuration file to? This is to confirm whether or not it's a local Python disk write permissions issue (which may indicate a problem with your Python installation).

semuadmin, 2024-01-08:
Hi @jxl77 Are you able to provide any further diagnostics on this, as per previous response?

jxl77, 2024-01-08:
Thanks for your concern. 

Since my first trys, I have decided to upgrade my system  to the latest Somona. 
I have also cleaned my python installation (there were 4 versions installed by different manners) 
Pygpsclient is now working properly. 

The bad thing is that I can no longer reproduce the bug which could have originated from  various sources. 

Thanks for your efforts.

semuadmin, 2024-01-08:
> Thanks for your concern.
>
> Since my first trys, I have decided to upgrade my system to the latest Somona. I have also cleaned my python installation (there were 4 versions installed by different [...]

OK glad it's sorted. Are you happy to close the issue now?

jxl77, 2024-01-08:
yes, we can close it

closed 2024-01-08 by semuadmin