semuconsulting/PyGPSClient #109: PIL.Image has no attribute ANTIALIAS

japs386, 2024-01-22:
**Describe the bug**

AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'

**Additional context**

pygnssutils-1.0.17, but all versions will see the error.

The problem and the solution is described here:
https://stackoverflow.com/questions/76616042/attributeerror-module-pil-image-has-no-attribute-antialias

The following change in site-packages/pygpsclient/ solves the problem:
Out:           Image.open(IMG_WORLD).resize((w, h), Image.ANTIALIAS)
In:              Image.open(IMG_WORLD).resize((w, h), Image.LANCZOS)

Problem started with upgrade from Pillow-9.5.0 to pillow-10.2.0
ANTIALIAS was removed in Pillow 10.0.0 (after being deprecated through many previous versions).

japs386, 2024-01-22:
PyGPSClient: 1.3.25

[japs@japspc ~]$ pygpsclient
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib64/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/usr/local/lib/python3.9/site-packages/pygpsclient/app.py", line 556, in on_gnss_read
    self.process_data(raw_data, parsed_data)
  File "/usr/local/lib/python3.9/site-packages/pygpsclient/app.py", line 669, in process_data
    frm.update_frame()
  File "/usr/local/lib/python3.9/site-packages/pygpsclient/map_frame.py", line 108, in update_frame
    self._draw_static_map(lat, lon)
  File "/usr/local/lib/python3.9/site-packages/pygpsclient/map_frame.py", line 128, in _draw_static_map
    Image.open(IMG_WORLD).resize((w, h), Image.ANTIALIAS)
AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'

semuadmin, 2024-01-22:
You’re using a very old version - the issue you’re referring to was addressed back in 1.3.28. I strongly recommend upgrading to the latest version 1.4.8. 

```shell
python3 -m pip install —upgrade pygpsclient
```

japs386, 2024-01-22:
I just did a few minutes ago, but pygpsclient/map_frame.py still contains 
Image.open(IMG_WORLD).resize((w, h))  # , Image.ANTIALIAS)
It may give rise to some confusion. Anyway problem solved.

semuadmin, 2024-01-22:
Not sure why a comment would cause confusion, but happy to remove it in next release.

Assuming you’ve upgraded, are you happy to close the issue?

closed 2024-01-22 by japs386