semuconsulting/PyGPSClient #110: Why the advanced configuratinos are hidden for socket server mode?

asukiaaa, 2024-01-29:
Thank you for sharing useful projects.
I want to send information to some caster with using socket server mode with fixing position.
However, the menu about `Configure Base Station` is hidden for that mode.
Why?
https://github.com/semuconsulting/PyGPSClient/blob/1a44663f8fbba64b15994ba21133a2a345eaaf79/src/pygpsclient/serverconfig_frame.py#L545

semuadmin, 2024-01-29:
Hi @asukiaaa

So "Socket Server" and "NTRIP Caster" are two completely different modes, [as described in the README](https://github.com/semuconsulting/PyGPSClient#socketserver). The "Configure Base Station" options only apply to "NTRIP Caster" mode, not to "Socket Server" mode. This is expected behaviour.

asukiaaa, 2024-01-30:
Hi @semuadmin 

Thank you for the response.

> So "Socket Server" and "NTRIP Caster" are two completely different modes

I taught that "Socket Server" is server for ntrip.
Then, is there any way to build server of ntrip to send base station data to some ntrip caster like rtk2go?

semuadmin, 2024-01-30:
HI @asukiaaa 

PyGPSClient's "Socket Server" mode simply transmits the receivers raw data stream (NMEA, UBX or RTCM) over a TCP socket to any socket client capable of parsing such data. It does _NOT_ transmit the NTRIP protocol or act as an NTRIP caster - it's typically used to monitor the raw output from a remote receiver over a local area network.

PyGPSClient's "NTRIP Caster" mode takes a raw RTCM datastream from a receiver configured as a 'base station', wraps this in the NTRIP protocol and broadcasts this over HTTP. It essentially provides the same functionality as a public NTRIP service like rtk2go (_i.e. you can connect any NTRIP client to it_), but is only really intended for evaluation, testing and diagnostic purposes.

If you want to use rtk2go as your caster, you'll need follow their [registration procedure](http://rtk2go.com/sample-page/new-reservation/). rtk2go recommend that you use [SNIP software](https://www.use-snip.com/) for production purposes, but I believe there's nothing stopping you using PyGPSClient's NTRIP caster as a 'Push-Out' feed. Again, refer to rtk2go's  documentation for the relevant procedures.