semuconsulting/PyGPSClient #123: Support Ublox NTRIP Client (No MQTT) SPARTN format

jptalledo10, 2024-03-16:
# PyGPSClient Feature Request Template

**NB**: Please raise any feature requests or queries in the [pygpsclient Discussions Channels](https://github.com/semuconsulting/pygpsclient/discussions) in the first instance.

**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is e.g. I'd like to be able to do this [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

**Would you be willing to contribute a test device?**

If the request relates to a specific u-blox device that is not currently supported, would you be 
willing to contribute a device to the project for testing purposes?

**Additional context**

Add any other context about the feature request here.

semuadmin, 2024-03-16:
Hi @jptalledo10 ,

PyGPSClient can already parse (_but not yet fully decode_) SPARTN data from MQTT and L-Band sources, so in theory there should be no major difficulty in parsing SPARTN data from an NTRIP (i.e. HTTP) source.

The main issue is that, while the SPARTN protocol is ostensibly in the public domain, the documentation is (_in my experience_) relatively poor and u-blox's SPARTN services all require a paid subscription in order to decrypt the raw data, making it extremely difficult for an unpaid and unsponsored volunteer project like PyGPSClient to develop, test and implement it successfully. SPARTN is basically a money-spinner for u-blox.

The NTRIP RTCM3 protocol, on the other hand, is well documented, unencrypted and a great many NTRIP sources are available free of charge from public domain service providers like rtk2go.com and euref-ip.com.

u-blox have shown little interest in supporting open source projects like PyGPSClient and certainly no interest in assisting with subscription fees or testing, so at this point I don't feel greatly motivated to implement new SPARTN functionality in support of u-blox's revenue stream.

As things stand, the [pyspartn](https://github.com/semuconsulting/pyspartn) library (_which PyGPSClient uses to parse, decrypt and decode SPARTN data_) is currently still in Alpha and cannot yet fully decode all SPARTN payloads. I suspect the outstanding issues are probably easily resolved (_they're almost certainly a document interpretation problem rather than a decryption problem_), but I've simply not had the time or frankly the incentive to look into them in any depth.

Having said this, contributions of all kinds are always welcome so if anyone out there is in a position to assist with the implementation of such functionality, I'd be more than happy to incorporate it.

jptalledo10, 2024-03-16:
The good news is that Ntrip Spartn is not encrypted and I can share a developer account for debugging/testing if needed.  Happy to discuss further.

semuadmin, 2024-03-17:
> The good news is that Ntrip Spartn is not encrypted and I can share a developer account for debugging/testing if needed. Happy to discuss further.

That would certainly be helpful - appreciate the offer. In point of fact, you don't actually need to _decrypt_ SPARTN data in order to send it to an RTK-compatible GNSS receiver like the ZED-F9P - you can simply send the raw binary datastream along with the relevant decription keys and the receiver will happily decrypt it itself. Native SPARTN decryption in [pyspartn](https://github.com/semuconsulting/pyspartn) is provided more as a diagnostic and academic service to end users.

I've been reviewing the PointPerfect service description at https://developer.thingstream.io/guides/location-services/pointperfect-ntrip-distribution and it all looks relatively straightforward, particularly as it appears to support the standard NTRIP HTTP port 2101 in addition to an encrypted HTTPS (TLS) service on port 443.

Supporting the service would require the following changes:
1. (_necessary_) Enhancements to the [pygnssutils](https://github.com/semuconsulting/pygnssutils) library - specifically the [gnssntripclient](https://github.com/semuconsulting/pygnssutils/blob/main/src/pygnssutils/gnssntripclient.py) class. It should be a simple matter to add a flag to signify either RTCM3 or SPARTN protocol, and then assign either [pyrtcm.RTCMReader](https://github.com/semuconsulting/pyrtcm) or [pyspartn.SPARTNReader](https://github.com/semuconsulting/pyspartn) to parse the data into individual RTCMMessage or SPARTNMessage objects and forward the serialised objects to the receiver.
2. (_optional_) For HTTPS support, I may also need to add a parameter which points to the location of the necessary PointPerfect TLS certificates, assuming these are not already supported natively by the Python `certifi` library via its default `cacerts.pem` TLS certificate registry.
3. (_optional_) enhancements to the `pyspartn` library to fix the remaining SPARTN decode issues - something that's been on my backlog for a while anyway. Having access to a developer account would certainly facilitate this.

I'm moving this issue to the [PyGPSClient Ideas and Feature Requests Discussion Forum](https://github.com/semuconsulting/PyGPSClient/discussions/categories/ideas-and-feature-requests) where we can continue the discussion, if that's OK with you.

Thanks again for the suggestion and any assistance you can offer.