semuconsulting/PyGPSClient #177: Update not possible?

Merga189, 2025-03-14:
**Describe the bug**

Trying to update two installations on different desktop machines (DEBIAN bookworm) failed.

[image]

Starting pygpsclient from the terminal and trying to update, It does not show any error messages ....

In 2013 on both installations I recorded my terminal commands and I did one installation using pip and the other installation used pipx.(I am just a normal Linux user and not an expert on python.)

Having a closer look into those installation protocols I see in the pip installation those warning 

> WARNING: The scripts pyserial-miniterm and pyserial-ports are installed in '/home/kalle/.local/bin' which is not on PATH.
>   Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
>   WARNING: The scripts gnssdump, gnssmqttclient, gnssntripclient, gnssserver, ubxload, ubxsave and ubxsetrate are installed in '/home/kalle/.local/bin' which is not on PATH.
>   Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
>   WARNING: The script pygpsclient is installed in '/home/kalle/.local/bin' which is not on PATH.
>   Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
> Successfully installed cffi-1.15.1 cryptography-41.0.3 paho-mqtt-1.6.1 pycparser-2.21 pygnssutils-1.0.15 pygpsclient-1.4.3 pynmeagps-1.0.28 pyrtcm-1.0.12 pyserial-3.5 pyspartn-0.1.10 pyubx2-1.2.30
> 

and in the pipx installation 

> ~$ pipx install pygpsclient
>   
>   installed package pygpsclient 1.4.3, installed using Python 3.11.2
>   These apps are now globally available
>     - pygpsclient
> ⚠️  Note: '/home/kalle/.local/bin' is not on your PATH environment variable. These apps will not
>     be globally accessible until your PATH is updated. Run `pipx ensurepath` to automatically add
>     it, or manually modify your PATH in your shell's config file (i.e. ~/.bashrc).
> done! ✨ 🌟 ✨

Both installations work as expected, only the update is failing. Could the path thing be the cause? Is the update feature is supposed to work?

br

PS. PyGPSClient as appimage would be great!

semuadmin, 2025-03-14:
Hi @Merga189 

I've just tried the GUI update facility on my own Debian Bookworm platform and it appears to be working OK, and I've had no other reports of issues on Linux, so I'm guessing this may be a local issue:

[image]

Some observations and queries...

The Installation section of the README does make the following recommendations:

[image]

The recommended installation command (which also serves for upgrades) is:

```shell
python3 -m pip install --upgrade pygpsclient
```

1. Does the command above work for you as a terminal command (as opposed to via the GUI)?
2. How exactly did you install Python on your platform, or are you using the 'default' distribution version?
3. How exactly did you install PyGPSClient originally? What specific command did you use, and did you install into a virtual environment?
4. Is there anything stopping you adding the Python user binaries folder  `/home/kalle/.local/bin` to your PATH, via BASHRC or equivalent for your shell?

semuadmin, 2025-03-14:
> PS. PyGPSClient as appimage would be great!

Acknowledged, and there is a dedicated PyPi utility [python-appimage](https://python-appimage.readthedocs.io/en/latest/apps/) for this purpose, but in the past I've found it unreliable for tkinter (GUI) application packaging on all Linux distributions. You're welcome to try this yourself on your particular platform.

One of the key aims here is to provide a simple installation mechanism that works for _all_ common platforms, including Windows, MacOS and the various flavours of Linux. In my experience, Python's native pip package manager serves this purpose pretty well _provided_ the recommendation installation procedures are followed.

Adding platform- or distribution-specific installation packages like AppImage, Snap, Flatpack, MSI, DMG, etc. etc. greatly increases the (_unpaid voluntary!_) support overhead. I'd prefer to avoid going down this road unless there is overwhelming demand.

Merga189, 2025-03-14:
First of all, thank you for having a look.

Debian stable at the moment comes with python 3.11.2 and trixi/testing with python 3.13.2. In Debian discussions it is always pointed to use the Debian resources. I personally would not like to use the external python source ...

If I have to do a fresh installation of PyGPSClient-1.5.3, I'll do it in an Debian trixi/testing environment, as the official python is also 3.13.2

> It is strongly recommended that** the Python 3 [binaries](#binaries) and site_packages directories are included in your PATH (*most standard Python 3 installation packages will do this automatically if you select the 'Add to PATH' option during installation*).

Might be the problem here, as both installations in 9/2023 worked, I did not into those PATH warnings.

TODAY added the path to .bashrc in the pipx installation. Still update failed!

I add my installation log here and hope, it is not to much, if you have a short look. Always try to make it readable for me (and others), if I have to redo an installation later.

Page 1 is the pip installation
Page 4 is the pipx installation
Page 15 is adding path with the pipx install.

[Pygpsclient - Installation with python and pipx.odt](https://github.com/user-attachments/files/19253238/Pygpsclient.-.Installation.with.python.and.pipx.odt)

Just tried this on both machines here

```
~/.local/bin$ python3 -m pip install --upgrade pygpsclient
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

br

semuadmin, 2025-03-14:
@Merga189 

OK apologies if you're already familar with this, but with Python installations you essentially have two options:

1. You can install every application into the platform's common `site-packages` folder directly.
2. You can install each individual application into its own segregated ['virtual environment'](https://www.geeksforgeeks.org/python-virtual-environment/) with its own `site-packages` folder. If your Python environment is 'externally managed' (which is often the case with default Linux distros),  you are effectively required to use this option.

The advantage of (2) is that it avoids potential conflicts between different applications' package requirements e.g. if one app requires `pyserial<=3.4` and another requires `pyserial==3.5`, installing each app into its own virtual environment means they can happily coexist on the same platform. The downside is that _you have to activate each virtual environment before you can invoke its application_.

To install and run PyGPSClient in a virtual environment, as per the README:

```shell
python3 -m pip install --user --upgrade virtualenv
python3 -m virtualenv env
source env/bin/activate
python3 -m pip install --upgrade pygpsclient
pygpsclient
```

To deactivate the virtual environment:

```shell
deactivate
```

To rerun the application from the virtual environment:

```shell
source env/bin/activate
pygpsclient
```

(_FYI pipx is essentially a wrapper around the pip and virtualenv tools which does this under the covers - I personally prefer to run the native commands, but that's just me_)

Unfortunately, the Update facility invoked from PyGPSClient's About dialog does not currently cater for virtual environments - it assumes the application has been installed into the common site-packages folder. If you've installed PyGPSClient into a virtual environment, you can still use the terminal commands above to update it, but the GUI facility will not work.

If PyGPSClient is the only significant Python app you're intending to run on your platform, there is no real benefit in using a virtual environment. In these circumstances, you could 'cheat' and simply remove or rename the `/usr/lib/python3.x/EXTERNALLY-MANAGED` marker, which will allow you to install PyGPSClient into the common area as per (1). You should then find the GUI update faciltiy works fine.

I would strongly advise AGAINST using the `--break-system-packages` pip directive, as this can cause all sorts of downstream problems which may not be apparent at the time of installation.

I'll put an action on my backlog to enhance the GUI update facility, but in the meantime I'd either:
a) remove the EXTERNALLY-MANAGED marker and install the app into the common area (option 1).
or
b) install into a virtual environment (option 2, using native pip/virtualenv or, if you prefer, pipx) and use terminal commands to update the application.

Merga189, 2025-03-15:
Pardon me, if somehow I insulted you, but English is not my natural language.

In 2023 when I installed PyGPSClient I was happy, with my very limited experience, that I got it working.

But you can not tell me, what went wrong, so an update is not possible with both installations?

Summing up your last answer, you emphasize, I use the virtual installation, can not use then GUI, but can use the command line to update PyGPSClient

I'll try a new installation on Debian testing/trixie.

If questions come up, I hope I can put these here ...

Think, you can close this here.

br from the Rheinland ;-)

semuadmin, 2025-03-15:
> Pardon me, if somehow I insulted you, but English is not my natural language.

Not a problem, and absolutely no insult was inferred!

A fix for this issue is in hand, so watch this space - the comments above were purely intended as interim measures that you could take in the meantime.

If you have any further queries, don't hesitate to post them to the [Discussions](https://github.com/semuconsulting/PyGPSClient/discussions/categories/q-a) forum in the first instance.

Tschüss!

Merga189, 2025-03-15:
I'll give it a try in Debian trixie (running in virtualbox here) and if questions on my side come up, I come back in discussion forum ;-)

Thank you for your patience with an old guy and the nice conversation.

See you!

PR #180 (2025-03-17): RC 1.5.4
# PyGPSClient Pull Request Template

## Description

1. Fix issue with GUI update facility not working for virtual environments.

Fixes #177

## Testing

Please test all changes, however trivial, against the supplied pytest suite `tests/test_*.py`. Please describe any test cases you have amended or added to this suite to maintain >= 99% code coverage.

- x] Tested for install and update using pip and pipx

## Checklist:

- [x] I agree to abide by the code of conduct (see [CODE_OF_CONDUCT.md](https://github.com/semuconsulting/pygpsclient/blob/master/CODE_OF_CONDUCT.md)).
- [x] My code  [... cut]
commit 280f3f20 (2025-03-17): update about_dialog Fixes #177
closed 2025-03-15 by Merga189